elupus/esphome-nibe #2: LilyGo T-CAN485 not working

Tommatheussen, 2022-10-03:
I'm currently trying to get this board working with your repo:
https://github.com/Xinyuan-LilyGO/T-CAN485

I can get ESPHome up and running, and it seems it does go through the setup process, as it's logging the config to the console:
```
[19:26:33][C][nibegw:122]: NibeGw
[19:26:33][C][nibegw:128]:  Target: xx:10090
[19:26:33][C][nibegw:132]:  Source: xx
[19:26:33][C][nibegw:134]:  Read Port: 9999
[19:26:33][C][nibegw:135]:  Write Port: 10000
```
The problem is that when I enable MODBUS on my heatpump (F1245PC) it starts to error out, I've tried switching the A and B sides, as well as switching the pins on the ESPHome config file, nothing seems to work.
I've tried some debugging and I found that the serial connection doesn't get started. In fact, the log from [NibeGwComponent::setup](https://github.com/elupus/esphome-nibe/blob/6796622239b71a35493067f340f4e152cd6b2877/components/nibegw/NibeGwComponent.cpp#L113) never got logged so it seems the setup never gets called?

Tommatheussen, 2022-10-04:
I will try some more tests when I get home, mainly directly connecting the ESPHome node to my machine, to make sure I don't miss any logs during boot/startup.

It looks like the component does get set up properly, it's logging the receiving UDP packets when trying to set up from HA

elupus, 2022-10-04:
You wont see setup logs unless you have logs over serial port during a reset.

Are you getting any logged traffic?

elupus, 2022-10-04:
Ive not tested with an rs485 card that needs enable pins to work. So might need some adaptions.

elupus, 2022-10-04:
So you need to make sure the enable pin is set by other esphome means, and the SE pin should be the direction pin.

Tommatheussen, 2022-10-04:
I have indeed manually set the enable pin, but also the SE pin. According to the docs of the chip used, when it's pulled high during setup, that enabled auto direction mode. I'll keep you updated with my findings when using serial logging.

So far I was not able to get any logged traffic (from the RS485/modbus connection), will try with the SE pin as direction pin tonight.

Tommatheussen, 2022-10-04:
I got it working, indeed playing with the SE and EN pins was the right move. I actually tried that before, but was staring myself blind on the 'setup' phase.

In case you want to add this to the documentation, I've used these output pins to be pulled high:
```
output:
  - platform: gpio
    id: ENABLE_PIN
    pin:
      number: GPIO19
      inverted: true
  - platform: gpio
    id: SE_PIN
    pin:
      number: GPIO17
      inverted: true
```

elupus, 2022-10-04:
Since that is quite a nice board, perhaps you could document an example setup for that?

Tommatheussen, 2022-10-04:
Sure thing, I'll tweak some more things and play around a bit and send a PR with a full example.
Nice work BTW :)

closed 2022-10-06 by elupus

humbeecc, 2022-10-16:
The lilygo-t-can485.yaml works for me without the "board_build.partitions: huge_app.csv" option. 
Without it, this allows ota updates.
Is there a reason why that option is needed?

elupus, 2022-10-16:
I wondered the same. So lets remove it if it works without it.

Tommatheussen, 2022-10-16:
Never actually tested the final config without, I took it over from their example docs. If it works without then I'm fine with removing it.