elupus/esphome-nibe #8: Problem with compiling

Manhem, 2022-10-20:
Hi.
I have now made a PCB for this project an today I was trying to program my ESP8266.
For the declaration of the pin I tryed both GPIO1(TX), GPIO3(RX) and GPIO5(Direction).
I also tryed with TX, RX and D1.
Get the same error everytime.

I have used your example anf only change the pins.

Any thing I'm doing wrong?
I get this error:

Compiling /data/nibe-f1145/.pioenvs/nibe-f1145/src/esphome/components/nibegw/NibeGw.cpp.o
src/esphome/components/nibegw/NibeGw.cpp: In member function 'void NibeGw::connect()':
src/esphome/components/nibegw/NibeGw.cpp:52:38: error: invalid conversion from 'int' to 'SerialMode' [-fpermissive]
   52 |       RS485->begin(9600, SERIAL_8N1, RS485RxPin, RS485TxPin);
      |                                      ^~~~~~~~~~
      |                                      |
      |                                      int
In file included from /data/cache/platformio/packages/framework-arduinoespressif8266/cores/esp8266/Arduino.h:288,
                 from src/esphome/components/nibegw/NibeGw.h:39,
                 from src/esphome/components/nibegw/NibeGw.cpp:19:
/data/cache/platformio/packages/framework-arduinoespressif8266/cores/esp8266/HardwareSerial.h:87:68: note:   initializing argument 3 of 'void HardwareSerial::begin(long unsigned int, SerialConfig, SerialMode, uint8_t)'
   87 |     void begin(unsigned long baud, SerialConfig config, SerialMode mode, uint8_t tx_pin)
      |                                                         ~~~~~~~~~~~^~~~
Compiling /data/nibe-f1145/.pioenvs/nibe-f1145/src/esphome/components/nibegw/NibeGwComponent.cpp.o
*** [/data/nibe-f1145/.pioenvs/nibe-f1145/src/esphome/components/nibegw/NibeGw.cpp.o] Error 1
In file included from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/stl_tree.h:65,
                 from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/set:60,
                 from src/esphome/components/nibegw/NibeGwComponent.h:3,
                 from src/esphome/components/nibegw/NibeGwComponent.cpp:1:
/data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/stl_function.h: In instantiation of 'constexpr bool std::less<_Tp>::operator()(const _Tp&, const _Tp&) const [with _Tp = IPAddress]':
/data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/stl_tree.h:2568:33:   required from 'std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::const_iterator std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::find(const _Key&) const [with _Key = IPAddress; _Val = IPAddress; _KeyOfValue = std::_Identity<IPAddress>; _Compare = std::less<IPAddress>; _Alloc = std::allocator<IPAddress>; std::_Rb_tree<_Key, _Val, _KeyOfValue, _Compare, _Alloc>::const_iterator = std::_Rb_tree<IPAddress, IPAddress, std::_Identity<IPAddress>, std::less<IPAddress>, std::allocator<IPAddress> >::const_iterator]'
/data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/stl_set.h:749:25:   required from 'std::set<_Key, _Compare, _Alloc>::size_type std::set<_Key, _Compare, _Alloc>::count(const key_type&) const [with _Key = IPAddress; _Compare = std::less<IPAddress>; _Alloc = std::allocator<IPAddress>; std::set<_Key, _Compare, _Alloc>::size_type = unsigned int; std::set<_Key, _Compare, _Alloc>::key_type = IPAddress]'
src/esphome/components/nibegw/NibeGwComponent.cpp:57:69:   required from here
/data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/10.3.0/bits/stl_function.h:386:20: error: ambiguous overload for 'operator<' (operand types are 'const IPAddress' and 'const IPAddress')
  386 |       { return __x < __y; }
      |                ~~~~^~~~~

elupus, 2022-10-20:
The esp8266 target seem to need some changes to work with its serial driver.

Also seem to need some adjustments for compiler. I might be able to fix it. But not right away.

elupus, 2022-10-20:
So I got compilation working. It will ignore the rx/tx pins you set and use the fixed ones for the uart. I think you also need to disable logging to uart completely so you don't start writing log information to the uart used for rs485

closed 2022-10-20 by elupus