elupus/esphome-nibe #9: Problem with programing ESP8266

Manhem, 2022-10-21:
I still have the same problem like before.
It seems that when I press install it don't download the new version (I think).

elupus, 2022-10-21:
I think you need to do a clean first.

Manhem, 2022-10-21:
So far so good but....
Still have one problem when compiling.
I use you full complete config but get this error:

nibegw: [source /config/esphome/nibe-f1145.yaml:44]
  dir_pin: GPIO5
  rx_pin: GPIO1
  tx_pin: GPIO3
  uart_id: 1
  udp: 
    target: 
      - ip: 192.168.0.31
        port: 9999
    source: 
    read_port: 9999
    write_port: 10000
  acknowledge: 
    - MODBUS40
    
    not a valid value.
    - RMU40_4

elupus, 2022-10-21:
Should have been RMU40_S4. I have corrected readme.

Manhem, 2022-10-21:
Now it works so far.
Just one more question for now.
Do the ESP send the ACK to the heatpump or do the ACK comes from Home Assistant?

elupus, 2022-10-21:
The ESP always acks, so the pump think the accessory is there even if HA restarts.

Manhem, 2022-10-21:
Thanks.
I have now downloaded the software and it works, but I can't find it in HA.
Are there something special I must do to get it to work?

elupus, 2022-10-21:
You have to add the nibe heat pump integration. Its not exposing any sensors directly.

Manhem, 2022-10-21:
Thanks.
I will now try it and if I get more problems I will get back to you later.

Manhem, 2022-10-22:
Have now tried a lot of things but the pump is going into alarm state every time,
I have an other equipment that works well.
Don't know wjhat to do next.
Maybe something wrong with the enable pin declaration.
Tried both with D1 and GPIO5, same result. (same thing different name)
Maybe I need to connect my analyser to see what happends.
Any surgestions?

elupus, 2022-10-22:
Could be enable pin declaration, could be logging using the using the uart on the esp. Could be something else with code on that arch since i have only run on esp32.

Do you get data logged from the esp when you view its debug log. That way you know if you have working reception. But if you have a analyser that is likely good.

Manhem, 2022-10-22:
Is it possible to invert the enable pin?
I don't know if it's normaly high or normaly low in you program.

elupus, 2022-10-22:
You can enable invert on pins in standard esphome pin config i think.

Manhem, 2022-10-22:
Thanks.
I shall try that.

elupus, 2022-10-22:
I have updated the component to allow invert on the dir_pin. You also need to move the uart config to a separate block since it's now using native uart block of esphome. Check readme.

Manhem, 2022-10-23:
It works so far.
You have one small error in your config file at Github.
nibegw:
  dir_pin:
    number: GPIO4
    invert: false
SHALL BE:
    inverted: false

Do you have a description over all the ENTITETER in Home Assistant?
I only have more then 1500 entiteter to choose from.

elupus, 2022-10-23:
Nice! You can find some info in nibe modbus manager, we have copied them here too; https://github.com/yozik04/nibe/tree/master/nibe/data

I will try to add some higher level abstractions too, like climate: https://github.com/home-assistant/core/pull/78952 water_heater: https://github.com/home-assistant/core/pull/79099

But grouping and info is a bit sparse.

Manhem, 2022-10-23:
I have made a pcb with ESP8266, level convert and RS485 converter on it.
Shall I send you a copy of the schema?

elupus, 2022-10-23:
Sure! We can add it to our examples.

Manhem, 2022-10-24:
Hi.
I send you the schema I made and are now using.
Also the picture of the equipment I use.
Do what you want with it.

I think you need to do a better description both on esphome-nibe and Home Assistant Nibe.
There are little to many questionmarks about how to get it to work.

I will now test it and see how it works over time.
[image]
[image]
[image]
[Nibe Fxxx Heatpump.pdf](https://github.com/elupus/esphome-nibe/files/9851344/Nibe.Fxxx.Heatpump.pdf)

elupus, 2022-10-24:
Feel free to update the descriptions. I did this for my own use, with best effort support for others. I will try to add the schematics somehow.

elupus, 2022-11-25:
@Manhem could you testout https://github.com/elupus/esphome-nibe/pull/20 on your esp8266? I would like to switch to that of possible. But preferably not break 8266 support.

Manhem, 2022-11-26:
I can update ESPhome and download your software and after that 
I can do a clean installation and see what's happend.
You have put everything in your software, so it's just for me to install it?

elupus, 2022-11-26:
I merged the pull request. It seems to work fine on ESP32. So if you update and it fail. you should be able to fix your external component to the old version v2.0.3. The new version I want you to test is v2.1.0. Verify that that is what you get. You need to do a clean build.

Manhem, 2022-11-27:
I have now done a new installation, but where can I find what version I have installed??

elupus, 2022-11-27:
You have to check while building. It should show in the log.

Manhem, 2022-11-27:
What shall I look for now when I have installed the new software?

I have done a check in the log file and found this:
This is the only things that starts with 2. exept for my IP.

Library Manager: Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
INFO Installing esphome/ESPAsyncWebServer-esphome @ 2.1.0
Unpacking  [####################################]  100%
Library Manager: ESPAsyncWebServer-esphome@2.1.0 has been installed!
INFO ESPAsyncWebServer-esphome@2.1.0 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Library Manager: Installing ESPAsyncUDP

Dependency Graph
|-- ESPAsyncTCP-esphome @ 1.2.3
|-- ESPAsyncWebServer-esphome @ 2.1.0
|   |-- ESPAsyncTCP-esphome @ 1.2.3
|   |-- Hash @ 1.0
|   |-- ESP8266WiFi @ 1.0

elupus, 2022-11-27:
Looks alright. It has installed ESPAsyncUDP which was the new feature.

Manhem, 2022-11-27:
So, what shall I look for.
Anything special?
Lost of connection, missing sensor values....

elupus, 2022-11-27:
Loss of communication mainly.

elupus, 2023-10-18:
Is this issue still relavant?

rriekki, 2023-11-03:
What programming tool is supposed to be used for this ESP32 Nibe project?
I use Arduino IDE and PlatformioIO, but the code is not completely suitable for either.

elupus, 2023-11-03:
I use vscode. I would suspect that is what the esphome project uses.