elupus/esphome-nibe #11: config with new uart section not working

ricerweb, 2022-10-25:
Hi, i had a working setup (esp32+max485) with the "old" config: 

```
nibegw:
  dir_pin: GPIO4
  rx_pin: GPIO3
  tx_pin: GPIO1
  uart_id: 1
  udp:
    target:
      - ip: 10.0.0.163
        port: 10090
    source:
      - 10.0.0.163
  acknowledge:
    - MODBUS40
  constants:
    - address: MODBUS40
      token: ACCESSORY
      data: [
            0x0A, # MODBUS version low
            0x00, # MODBUS version high
            0x01, # MODBUS address?
      ]
```

I tried to update it with the new uart section, but my heat pump is going constantly into an error. 

```
uart:
  rx_pin: GPIO3
  tx_pin: GPIO1
  baud_rate: 9600
nibegw:
  udp:
    target:
      - ip: 10.0.0.163
        port: 10090
    source:
      - 10.0.0.163
  acknowledge:
    - MODBUS40
  constants:
    - address: MODBUS40
      token: ACCESSORY
      data: [
            0x0A, # MODBUS version low
            0x00, # MODBUS version high
            0x01, # MODBUS address?
      ]
```

elupus, 2022-10-25:
Looks like you removed dir_pin from the nibe section?

ricerweb, 2022-10-25:
Yes, I tried it with the dir_pin and without. 
I use an max485 so the dir_pin should not be necessery, as mentioned in the new Readme?

elupus, 2022-10-25:
Strange. Im running with similar setup. Is this esp32 or esp8266?

ricerweb, 2022-10-25:
Yes, it is esp32. 

In the Log is 1 Warning: 

> [09:36:11][W][uart.arduino_esp32:162]:   You're using the same serial port for logging and the UART component. Please disable logging over the serial port by setting logger->baud_rate to 0.

elupus, 2022-10-25:
If its esp8266 i think you need to change logger (https://esphome.io/components/logger.html) component to a baud_rate of 0 to disable it from using the uart.

elupus, 2022-10-25:
Ok, so applies to the esp32 too.

ricerweb, 2022-10-25:
Thank you, got it working again. 
I had to use GPIO16 and GPIO17 and set the logger baud_rate to 0.

elupus, 2022-10-25:
That is indeed better, but would have been good to know if it worked on original pins with logger off.

closed 2022-10-25 by elupus

ricerweb, 2022-10-25:
With the original Pins (GPIO3 and GPIO1) and logger off --> not working