elupus/esphome-nibe #12: Problem with lilygo-t-can485.yaml

frankvandenhurk, 2022-10-25:
I recently bought a lilygo-t-can485. I use the example from https://github.com/elupus/esphome-nibe/tree/master/examples without any modification and I get the following error:

`INFO Reading configuration /config/nibegw.yaml...
Failed config

uart: [source /config/nibegw.yaml:41]
  
  'baud_rate' is a required option for [0].
  - rx_pin: GPIO21
    tx_pin: GPIO22
nibegw: [source /config/nibegw.yaml:47]
  
  Component nibegw requires component logger.
  udp: 
    target: 
      - ip: 192.168.255.254
        port: 10090
    source: 
      - 192.168.255.254
  acknowledge: 
    - MODBUS40
  constants: 
    - address: MODBUS40
      token: ACCESSORY
      data: 
        - 10
        - 0
        - 1`

Am I doing something wrong of is there another problem?

frankvandenhurk, 2022-10-25:
Just added 

`logger:`

and

`baud_rate: 9600` (under uart)

and now it is working. Maybe add then to the example?

elupus, 2022-10-25:
Was logger actually needed.

elupus, 2022-10-25:
Never mind. Yes it needs it.

closed 2022-11-01 by elupus

frankvandenhurk, 2022-11-01:
tnx!

timvdsm, 2023-01-13:
Hi @frankvandenhurk
I am also preparing a lilygo-t-can485 board.
Should this line be added as well? 
```
wifi:
  power_save_mode: none
```
As I don't see it in the example yaml and it was mentioned as a recommendation.
Your lilygo-t-can485 board is working good? No problems anymore with the heatpump going into alarm? 

And does the Nibe heatpump first need to be powered-off before connecting the lilygo-t-can485 board to it? Or can you connect the lilygo-t-can485 board to the Nibe heatpump when the heatpump is powered-on?

frankvandenhurk, 2023-01-14:
I don't have that in my config and it is working perfectly

I connected the data lines first, then the - and then the + (without turning the heatpump off)

Tommatheussen, 2023-01-14:
> Hi @frankvandenhurk I am also preparing a lilygo-t-can485 board. Should this line be added as well?
>
> ```
> wifi:
>   power_save_mode: none
> ```
>
> As I don't see it in the example yaml and it was mentioned [...]

It's not needed to add that line, BUT, I had issues with the heatpump going into alarm mode (see https://github.com/elupus/esphome-nibe/issues/21). I've recently added this line, because my heatpump is not at an ideal location for wifi stability, and so far (🤞) I haven't had any issues since, I'm giving it another week or so before I can confidently say it helps, but it's looking promising so far.

timvdsm, 2023-01-14:
> my heatpump is not at an ideal location for wifi stability,

Thanks for your reply. I am a little bit in the same situation. Did you putt the lilygo-t-can485 inside the heatpump? Or how did you mount it? Because I guess that's also not an ideal place for wifi. 
I do have a network cable at my heatpump, so the best solution for me would be to add an ethernet port to the lilygo-t-can485 . But I don't know if that's possible.

Tommatheussen, 2023-01-16:
Yep, it's inside of the heatpump itself, inside the little lid where the connection terminals are.