elupus/esphome-nibe #14: no response from nibeGW

Coffee0297, 2022-10-30:
Hello, thanks alot for this addon. im having a little issue, i can't get the homeassistant Nibe heatpump:https://rc.home-assistant.io/integrations/nibe_heatpump
To connect. it says my remote ip or read port is wrong, but im using all defaults. 
Im using HA beta to get latest fixes.

[image]
[image]
[image]
But as you can see the ESP get the udp from HA it seems..

elupus, 2022-10-30:
Ive had oddness (and heard others) with the default ports. Set it up on 10080, 10081, 10082 and try again.

I think there is some chromecast stuff on 9999

elupus, 2022-10-30:
Wait. Doesnt look like your pump is connected and have modbus40 enabled.

Coffee0297, 2022-10-30:
> 

the heatpump have modbus enabled, gan grab a pic. its a VVM310 and when i disconnect the esp the heatpump errors

Coffee0297, 2022-10-30:
[image]
[image]

Coffee0297, 2022-10-30:
and seems like even with difrent ports same issue. 

Here is the part of esphome 

```# Load nibe component
external_components:
  - source: github://elupus/esphome-nibe

# Configure uart that will be used
uart:
  rx_pin: GPIO16
  tx_pin: GPIO17
  baud_rate: 9600

# Configure NibeGW
nibegw:
  dir_pin:
        number: GPIO3
        inverted: false
  udp:
    # The target address(s) to send data to. May be a multicast address.
    target:
      - ip: 10.0.0.225
        port: 10080
    # List of source address to accept data from, may be empty for no filter
    source:
      - 192.168.16.130

    # Optional port this device will listen to to receive read requests. Defaults to 9999
    read_port: 10081

    # Optional port this device will listen to to receive write request. Defaults to 10000
    write_port: 10082

  acknowledge:
    - MODBUS40
    - RMU40_S4

  # Constant replies to certain requests cabe made
  constants:
    - address: MODBUS40
      token: ACCESSORY
      data: [
            0x0A, # MODBUS version low
            0x00, # MODBUS version high
            0x01, # MODBUS address?
      ]

# Accessory version response
    - address: RMU40_S4
      token: ACCESSORY
      data: [
            0xEE, # RMU ?
            0x03, # RMU version low
            0x01, # RMU version high
      ]

    # Unknown response that nibepi uses
    - address: RMU40_S4
      token: RMU_DATA
      command: RMU_WRITE
      data: [
            0x63,
            0x00,
      ]

    # Constant fixed temperature to avoid pump going into alarm.
    - address: RMU40_S4
      token: RMU_WRITE
      data: [
            0x06, # Temperature
            0x14, # degrees low
            0x00, # degrees hight
      ]

# Some helper functions to restart ESPHome from HA
button:
- platform: restart
  name: Nibegw Restart
- platform: safe_mode
  name: Nibegw Safe Mode Boot```

elupus, 2022-10-30:
Are you running in docker or some other virtualization system? If so you need to set up port forward.

Coffee0297, 2022-10-30:
> Are you running in docker or some other virtualization system? If so you need to set up port forward.

im running HA in a proxmox. but all other esp have 0 issues, but that may be bc default ports?

Coffee0297, 2022-10-30:
sadly port forward did not fix it either.. can i enable some deep debug to see what the esp recives on the ip? and if it sends anything

elupus, 2022-10-30:
Well, this is not standard esphome communication. It is nibegw communication.

You need to port forward the 9999/local listening port into your HA device if its not on a bridged network.

Yes, you can turn on debug with these entries for logger component in home assistant.

```
homeassistant.components.nibe_heatpump: debug
nibe: debug
```

The esphpme device can be increased too:
```
nivegw:
  debug: true
```

Coffee0297, 2022-10-31:
i tried to portforward no dice... ill try some more debugging. 
Also is it exspected that the install of this will dissable a outside heatpump? mine have not been running when the esphome was connected, but taking it off and the heatpump stated preheating. it seems to be the same rs485 but difrent ports so not sure if theres any correlation

elupus, 2022-11-01:
It should absolutely not affect any other pumps. That even more indicate something broken with the esp connection to the pump. For example th rs485 chip always in transmit mode.

What hw are you using?

Coffee0297, 2022-11-01:
Esp32 and a uart to rs485 
[image]
[image]
The odd thing is the heat pump says no error

elupus, 2022-11-01:
What type of rs485 chip is that?

elupus, 2022-11-01:
Ps. I would have expected a debug line in the esp log, every time it sends data over udp. So once wifi is up, you should be getting logs every second.

Coffee0297, 2022-11-01:
Max485 
Same type as I have:
https://www.amazon.com/ANMBEST-Transceiver-Arduino-Raspberry-Industrial-Control/dp/B088Q8TD4V

I apriciate all the help! 

Regarding the debug, i tried adding it but i only get a recive 6 packets in the log, nothing else. but would make sense if the Modbus is fucked..

elupus, 2022-11-01:
Also, you could look at the rs485 lines with you oscilloscope, there should be traffic there once connected.

elupus, 2022-11-01:
That chip needs you to control the enable pins correctly. It has not auto mode. So verify that it is connected properly and or inverted properly.

Coffee0297, 2022-11-01:
I will check thanks, but would the heat pump not go in error if wrong flow control pin? As it shuld not get the ACK\NAK from the esp? Just trying to understand :)

elupus, 2022-11-01:
I dont know. Maybe it hangs if bus is always busy/dominant.

Coffee0297, 2022-11-01:
So I have it connected to the right pin, but do you change the dit pin on send or recive? This is how mine expect
High to transmit low to recive. Mine is non inverted in the config.
 
[image]

elupus, 2022-11-01:
Seems correct. I would look at the bus and verify that you have trafic after the esp is connected.

elupus, 2022-11-01:
The A/B pins could potentially be swapped too. But I assumed you tested swapping those.

Coffee0297, 2022-11-01:
Yea i had to swap a/b to clear the error from the heat pump

elupus, 2022-11-01:
Weird. Re verify the logs. It should be printing things as it sends it out on udp.

Coffee0297, 2022-11-01:
i poked a bit arround with the scope, have the DIR pin been tested before? to me it seems like the esp recives the first com fine, then puts the dir high as it shuld, but the dir pin do not get pulled low after the TX is done...

elupus, 2022-11-01:
I know there are people using it. But, the way the esp pulls pins low is by high impedance pulldown i think. I think the other people have used an inverted pin.

Coffee0297, 2022-11-01:
Yea good point .. i have tried both inverted true and false, same end result .. may just get one with auto direction and be done with it. Do we get access to more than on the Nibe uplink? Using it currently and works great, tho no actual kw produced so can't calc a cop

elupus, 2022-11-02:
More 8nternal atate + faster rate. But not much more that is very usefull to be honest.

elupus, 2022-11-04:
GPIO3 is part of the debug uart. That might be why it behaves wrongly.

elupus, 2022-11-18:
I think this is resolved now. Please recheck with the new release.

closed 2022-12-17 by elupus