elupus/esphome-nibe #16: Problem with HA connection

kirilldem, 2022-11-04:
Hi elupus,

Thanks for this awesome software! I have bought the lilygo rs485 device to connect to my F730. I can connect to it from a separately hosted nibepi but having issues with HA connection. It is probbaly due to my target and source settings. Any ideas what I can change to connect?

I get the following error in HA:

> The model selected doesn't seem to support modbus40

[image]

It may well be that I have too much stuff in my ESP configuration as I combined your full configuration example and the lilygo example. This is my ESPhome configuration:

```
esphome:
  name: nibegw
  comment: Nibegw
  platform: ESP32
  board: esp32dev

# General ESPHome setup
api:

ota:
  password: !secret OTA_PASSWORD

logger:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true # Fast connect to connect to my hidden network

# Load nibe component
external_components:
  - source: github://elupus/esphome-nibe

# Set pins required for LilyGo T-CAN485 board
output:
  - platform: gpio
    id: ENABLE_PIN # Enable the chip
    pin:
      number: GPIO19
      inverted: true
  - platform: gpio
    id: SE_PIN # Enable autodirection
    pin:
      number: GPIO17
      inverted: true
  - platform: gpio
    id: ENABLE_5V_PIN # Enable 5V pin for RS485 chip
    pin:
      number: GPIO16
      inverted: true

# Configure uart that will be used
uart:
  rx_pin: GPIO21
  tx_pin: GPIO22
  baud_rate: 9600

# Configure NibeGW
nibegw:
  udp:    
    target:   # The target address(s) to send data to. May be a multicast address.
      - ip: 192.168.1.156 #NibePi
        port: 9999
      - ip: 192.168.1.13 #HA
#        port: 10090        

    # List of source address to accept data from, may be empty for no filter
    source:
      - 192.168.1.156   #NibePi
      - 192.168.1.13 #HA                                   
    read_port: 10000 # Optional port this device will listen to to receive read requests. Defaults to 9999                            
    write_port: 10001 # Optional port this device will listen to to receive write request. Defaults to 10000  

  acknowledge:
    - MODBUS40

    # Enable a dummy RMU40 accessory to receive updates
    # to certain registers faster. This should not be
    # enabled if you have an actual RMU40.
    - RMU40_S4

  # Constant replies to certain requests cabe made
  constants:
    - address: MODBUS40
      token: ACCESSORY
      data: [
            0x0A, # MODBUS version low
            0x00, # MODBUS version high
            0x01, # MODBUS address?
      ]
    # Accessory version response
    - address: RMU40_S4
      token: ACCESSORY
      data: [
            0xEE, # RMU ?
            0x03, # RMU version low
            0x01, # RMU version high
      ]
    # Unknown response that nibepi uses
    - address: RMU40_S4
      token: RMU_DATA
      command: RMU_WRITE
      data: [
            0x63,
            0x00,
      ]

    # Constant fixed temperature to avoid pump going into alarm.
    - address: RMU40_S4
      token: RMU_WRITE
      data: [
            0x06, # Temperature
            0x14, # degrees low
            0x00, # degrees hight
      ]      
# Some helper functions to restart ESPHome from HA
button:
- platform: restart
  name: Nibegw Restart
- platform: safe_mode
  name: Nibegw Safe Mode Boot
```

elupus, 2022-11-04:
That error looks to be missing parameter definition for that model in the communication library we use: https://github.com/yozik04/nibe

You can test with pump model F1155 just to verify if that is the issue. It will give you wrong list of available parameters, but i dont think you should get that error.

elupus, 2022-11-04:
Test with F750 it is closer and should work.

kirilldem, 2022-11-04:
Choosing F750 worked! Thank you!

closed 2022-11-05 by elupus