elupus/esphome-nibe #22: Heatpump unreachable with ESP8266

Manhem, 2022-12-18:
I have used the now firmware for three weeks now and everthing was working well until this night. (I was testpilot for ESP8266)
I have lost the connection to my ESP8266.
The connection to the pump is still there, so my heatpump is working.
The WiFi signal and ESPhome version is there but no values from the heatpump.
I have restart the ESP with the same result.

Logfile from ESP-home shows this:
INFO Reading configuration /config/esphome/nibe-f1145.yaml...
INFO Starting log output from 192.168.0.32 using esphome API
WARNING Can't connect to ESPHome API for 192.168.0.32: Error connecting to ('192.168.0.32', 6053): [Errno 113] Connect call failed ('192.168.0.32', 6053)
INFO Trying to reconnect to 192.168.0.32 in the background

I use Home Assistant:
Home Assistant 2022.12.1
Supervisor 2022.12.dev0901
Frontend-version: 20221208.0 - latest

elupus, 2022-12-18:
Do you get the logs okay network or direct connection over usb?

Manhem, 2022-12-18:
The answer I got was when I try to log wireless from ESPhome addon in Home Assistant

elupus, 2022-12-18:
Ok. Then its not finding the esphome on the network. Either it has broken down or its not finding wifi. I think you need to connect to it with a usb cable and get logs from it that way.

Manhem, 2022-12-18:
I have now reprogram the ESP for logging and the only thing I got was "start logging".
Then I reprogram the ESP again with baud: 0 again.
After that I got lots of text in Wireless logging:
12:01:51][C][wifi_signal.sensor:009]: WiFi Signal 'nibe-f1145_wifi_signal'
[12:01:51][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[12:01:51][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[12:01:51][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[12:01:51][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[12:01:52][D][nibegw:030]: UDP Packet with 6 bytes to send
[12:01:52][D][nibegw:030]: UDP Packet with 6 bytes to send
[12:01:53][D][nibegw:030]: UDP Packet with 6 bytes to send
[12:01:53][D][nibegw:030]: UDP Packet with 6 bytes to send
[12:01:53][D][nibegw:030]: UDP Packet with 86 bytes to send
[12:01:54][D][nibegw:030]: UDP Packet with 6 bytes to send
[12:01:54][D][nibegw:030]: UDP Packet with 6 bytes to send
[12:01:55][D][nibegw:030]: UDP Packet with 6 bytes to send
[12:01:55][D][nibegw:030]: UDP Packet with 6 bytes to send

But I don't get anything into Home Assistant, exept for WiFi signal and firmware version.

elupus, 2022-12-18:
Could your home assistant computer have changed its IP address?

Manhem, 2022-12-18:
Nope.
I have a static IP and all the other ESP's are working.
Also the IP for my Heatpump is static.

elupus, 2022-12-18:
Any errors in home assistant? Maybe something have blocked the port you use to receive data.

Manhem, 2022-12-18:
You mean port 9999 or 10000?
I why should it happend by it self?

elupus, 2022-12-18:
Im just guessing here. I did note that some chromecasts kept sending to 9999, so skipped those ports myself. That said, it should not be a problem since we filter the incoming port.

Could the ESP have changed its IP? Home assistant will filter on incoming address too.

Manhem, 2022-12-18:
The ESP has a static IP to.

elupus, 2022-12-18:
Did you configure home assistant with the IP of the esp or with the hostname?

Manhem, 2022-12-18:
I do everything in ESPhome addon:
esphome:
  name: ${devicename}

If this is what you mean?

elupus, 2022-12-18:
No, when you added the nibe integration.

Manhem, 2022-12-18:
I'm not really sure.
Where can I find that in HA?

Manhem, 2022-12-18:
I install Nibe addon again and the first test seems okej.
No I only need to redo everything.
Very strange....

elupus, 2022-12-18:
You should not need yo redo much i think.

Manhem, 2022-12-18:
Everything works again.
We must see if this is going to happend later. If it does, I think there can be a problem somewhere.
For now, we leave it like it is.

elupus, 2022-12-18:
Sounds good.

closed 2022-12-18 by elupus