elupus/esphome-nibe #24: Olimex esp32-poe

Berjastein, 2023-01-10:
Hi, im trying to configure Olimex esp32-poe, everything seems to be running, but i dont think its sending out udp packets, tried to point target after laptop and listen on wireshark, but there is nothing comming.

The pump is never going in alarm state, and modbus is activated on the pump, i i turn off the nibegw the pump goes in error, so i guess rs485 is working?

Im just using the wifi not the ethernet port.

Board:
https://www.olimex.com/Products/IoT/ESP32/ESP32-POE-ISO/open-source-hardware

RS485 addon for board:
https://www.olimex.com/Products/Modules/Interface/MOD-RS485/open-source-hardware

```
esphome:
  name: nibegw
  comment: Nibegw
  platform: ESP32
  board: esp32-poe

# General ESPHome setup
api:

ota:
  password: !secret OTA_PASSWORD

logger:

wifi:
  ssid: !secret WIFI_SSID
  password: !secret WIFI_PASSWORD
  fast_connect: true # Fast connect to connect to my hidden network

external_components:
  - source: 
      type: git
      url: https://github.com/elupus/esphome-nibe.git
    components: [ nibegw ]

uart:
  id: my_uart
  rx_pin: GPIO36
  tx_pin: GPIO4
  baud_rate: 9600

nibegw:
  udp:
    # The target address(s) to send data to. May also be multicast addresses.
    target:
      - ip: 192.168.86.171
        port: 10091

    # List of source address to accept read/write from, may be empty for no filter, but
    # this is not recommended.
    source:
      - 192.168.86.171

    # Optional port this device will listen to to receive read requests. Defaults to 9999
    read_port: 10091

    # Optional port this device will listen to to receive write request. Defaults to 10000
    write_port: 10092

  acknowledge:
    - MODBUS40

    # Enable a dummy RMU40 accessory to receive updates
    # to certain registers faster. This should not be
    # enabled if you have an actual RMU40.
    - RMU40_S4

  # Constant replies to certain requests can be made
  constants:
    - address: MODBUS40
      token: ACCESSORY
      data: [
            0x0A, # MODBUS version low
            0x00, # MODBUS version high
            0x02, # MODBUS address?
      ]

    # Accessory version response
    - address: RMU40_S4
      token: ACCESSORY
      data: [
            0xEE, # RMU ?
            0x03, # RMU version low
            0x01, # RMU version high
      ]

    # Unknown response that nibepi uses
    - address: RMU40_S4
      token: RMU_DATA
      command: RMU_WRITE
      data: [
            0x63,
            0x00,
      ]

    # Constant fixed temperature to avoid pump going into alarm.
    - address: RMU40_S4
      token: RMU_WRITE
      data: [
            0x06, # Temperature
            0x14, # degrees low
            0x00, # degrees hight
      ]
```

elupus, 2023-01-10:
Your config does look correct. Is it connecting to your wifi network? Can you ping it?

Have you look at the logs from the esp?

elupus, 2023-01-10:
You should be getting a log entry every time the pump speaks. You can add debug: true to the nibe section to get some more output. 

You dont seem to have set up a dir_pin? Are you sure that rs485 board support auto rx/tx switching?

Berjastein, 2023-01-10:
I am not sure, i cant find anything about it, do you know by looking at the diagram if it is or not?

Here is a link to schematic:
https://www.olimex.com/Products/Modules/Interface/MOD-RS485/resources/MOD-RS485-schematic.pdf

elupus, 2023-01-10:
That chip is not automatic. It needs to have its nRE and DE controlled. In your case they need to be tied together and controlled by the dir_pin setting.

At the moment the code cant drive both pins itself. Maybe some lambda could be used on the dir_pin setting but ive not looked at that.

Berjastein, 2023-01-10:
I made some changes on the print, and used GPIO 10 for dir_pin, now it is working, thanks for pointing me in the right direction :)

elupus, 2023-01-10:
Nice to hear!

closed 2023-01-10 by elupus