elupus/esphome-nibe #25: Component not found: nibegw

jonaswikstrom, 2023-01-28:
Hello

Very grateful for this initiative, thank you. This is possible an issue of documentation combined with missing knowledge about ESPHome. Simply trying the minimal config using edit in the the yaml editor for an ESP32 connected device.

Getting the error message:

ERROR Unable to import component nibegw:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 162, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/config/.esphome/external_components/b57125b4/components/nibegw/__init__.py", line 6, in <module>
    from esphome.const import (
ImportError: cannot import name 'CONF_DEBUG' from 'esphome.const' (/esphome/esphome/const.py)
Failed config

nibegw: [source /config/nibegateway.yaml:43]
  
  Component not found: nibegw.
  
  
What is failing?

elupus, 2023-01-28:
Sounds like something i need to fix. I think esphome have changed something.

elupus, 2023-01-28:
Try again

jonaswikstrom, 2023-01-28:
Still same problem...

jonaswikstrom, 2023-01-28:
Trying an update of my ESPHome server here.

jonaswikstrom, 2023-01-28:
Working now. My ESPServer were simply to old for the config to work. Everything compiles now.

closed 2023-01-28 by jonaswikstrom