elupus/esphome-nibe #30: Nibe heat pump integration - No entities shown

Sleazified, 2023-04-10:
Hello.

I have setup a lilygo-t board and connected it successfully to my nibe SMO40, i added the nibe heat pump integration, but i dont get any entities except for "Climate System S1".

[image]
 
[image]

the config part says +541 entities hidden. How can i activate some of these?

Cheers.

closed 2023-04-10 by Sleazified

rici44, 2023-08-14:
I got the same, how did you solve this?

Sleazified, 2023-08-14:
Hej Rici.

Go to the integration and activate the entities you want to activate, this
have to be done manually as there is too many entities in the integration.
And be wary of activating too many, then the integration crashes.

Den man. 14. aug. 2023 kl. 12.41 skrev Rici Mainly ***@***.***

>:

> I got the same, how did you solve this?

rici44, 2023-08-14:
Hi @Sleazified 

But when I try to change climate sensor I receive an error https://community.home-assistant.io/t/how-to-connect-to-nibe-heat-pump-without-the-cloud/381099/166?u=richi44

Sleazified, 2023-08-14:
> I cant do that either, sorry cant help with that :(
>
> Den man. 14. aug. 2023 kl. 13.30 skrev Rici Mainly <

i dont actually think you are supposed to be able to do that, you need to
activate the other entities and control instead.

rici44, 2023-08-14:
Hi @Sleazified. The climate integration is the most common way to control heating inside the house for the people inside. Its common thermostat which everybody is able to manage. I do not want to expose other settings of heat pump for everybody at home.

elupus, 2023-08-14:
The climate entity should work. But what are you trying to do? Changing mode is not really supported on the pump. It will always work against its degreeminutes, and use that to decide when to heat or not.

Target temperature should work.

rici44, 2023-08-14:
Hi @elupus Climate entity just is not working, do not know why. When I want to change target temperature it will show "Failed to call service climate/set_temperature. ‘set_temperature’ requires ‘hvac_mode’ when passing 'temperature and ‘hvac_mode’ is not already set to ‘heat’ or ‘cool’ "

https://community.home-assistant.io/t/how-to-connect-to-nibe-heat-pump-without-the-cloud/381099/169?u=richi44