elupus/esphome-nibe #31: Climate sensor

kirilldem, 2023-05-06:
Hello,

Does anyone have a solution on how to create a climate sensor and display it in HA as a generic thermostat? I see that the Nibe Uplink come with a climate component, but I would like to move away from the cloud component. 

Perhaps someone has a ready made solution you could share?

elupus, 2023-05-06:
For F series pumps, it already has one climate entity per nibe climate system.

You are not seeing them?

elupus, 2023-05-08:
Ok F730 must be missing some parameter needed. Will check.

Error does not seem related. Could ve corrupt data from device.

kirilldem, 2023-05-08:
Not sure if this matters but when I set up the integration in home
assistant I needed to choose F750 instead of F730. Otherwise it gave me an
error.

Is there a way to check for corruption issues on the GW?

elupus, 2023-10-18:
Was this resolved in HA?

kirilldem, 2023-10-18:
Still no climate entry from the gateway, only in the uplink integration. Still only one error in the logs:

``
Logger: nibe.nibe.connection.nibegw
Source: /usr/local/lib/python3.11/site-packages/nibe/connection/nibegw.py:220
First occurred: October 15, 2023 at 10:55:37 PM (4192 occurrences)
Last logged: 8:45:24 PM

Bad logger message: Unexpected exception during parsing packet data 'd0f281f88bff9af7d5ef94b6d1b4c09fec95e68fe187e8caf08bf68bf6' from ('192.168.1.230', 9999) ((ConstError(''),))
Bad logger message: Unexpected exception during parsing packet data 'd0f281f88bff9af7d5ef94b6d1b4c09fec95e68fe187e8caf08bf68ba785e08de89cf98ba993e8caadc8bce391f495f98de489eccef48ff28ff2' from ('192.168.1.226', 33188) ((ConstError(''),))
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/site-packages/nibe/connection/nibegw.py", line 182, in datagram_received
    msg = Response.parse(data)
          ^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/construct/core.py", line 288, in parse
    return self.parse_stream(io.BytesIO(data), **contextkw)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/construct/core.py", line 300, in parse_stream
    return self._parsereport(stream, context, "(parsing)")
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/construct/core.py", line 312, in _parsereport
    obj = self._parse(stream, context, path)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/construct/core.py", line 787, in _parse
    return self.parsefunc(stream, context)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "", line 58, in parseall
  File "", line 31, in parse_struct_1
  File "", line 21, in parse_const
construct.core.ConstError
``

tizianodeg, 2024-06-15:
@kirilldem can you check again, this should be working since HA 2024.6 version.

kirilldem, 2024-06-24:
> [@kirilldem](https://github.com/kirilldem) can you check again, this should be working since HA 2024.6 version.

I can confirm that I now have the climate system available in HA. Many thanks!

closed 2024-06-24 by kirilldem

Dutchstyler, 2024-11-13:
Just to break in...
Have the latest versions but can`t select a climate entity from the Nibe, what do I miss?
It`s a SMO40 with a F2040

I do have a control temperature entity.
[image]

[image]