elupus/esphome-nibe #34: detected dubious ownership in repository

frankvandenhurk, 2023-06-11:
I try to install a new version. It previously worked, but now I get:

```
NFO ESPHome 2023.5.5
INFO Reading configuration /config/nibegw.yaml...
Failed config

external_components: [source /config/nibegw.yaml:30]
  - 
    fatal: detected dubious ownership in repository at '/config/.esphome/external_components/b57125b4'
    To add an exception for this directory, call:
    
    	git config --global --add safe.directory /config/.esphome/external_components/b57125b4.
    source: github://elupus/esphome-nibe
```

It looks like checking has become more strict: https://github.blog/2022-04-18-highlights-from-git-2-36/#stricter-repository-ownership-checks

Is this cometing that can be fixed in the repo? Tnx in advance.

elupus, 2023-06-11:
No. That looks like an esphome issue. Maybe specific to the esphome addon in HA.

frankvandenhurk, 2023-06-11:
Ok, thanks for the quick reply. 

BTW: I don't use esphome as addon in HA but in a separate container.

closed 2023-06-11 by frankvandenhurk

elupus, 2023-06-11:
Looks like you have managed to change the user is running with at some point then.