elupus/esphome-nibe #36: Alarm Mode 251 every 15 Minutes

yannik1015, 2023-07-30:
I am running an Lilygo T-CAN485 with the config shown below on my Nibe F750 heatpump. However pretty much precisely every 15 minutes the heatpump goes into Error Mode (Code: 251). However when I connect wirelessly  to the T-CAN485 module to view the logs (`esphome logs config.yaml`) it stops getting these errors, but as soon as I stop the log the errors continue in their 15 minute interval.

Has anyone a idea what might causing this or how I could get additional information/logs to check further.

Config:
```
esphome:
  name: nibegw
  comment: Nibegw
  platform: ESP32
  board: esp32dev

# General ESPHome setup
# Enable Home Assistant API
api:
  password: <REDACTED>

ota:
  password: <REDACTED>

logger:
  level: DEBUG

wifi:
  ssid: <REDACTED>
  password: <REDACTED>
  fast_connect: true # Fast connect to connect to my hidden network
  power_save_mode: none

# Load nibe component
external_components:
  - source: github://elupus/esphome-nibe

# Set pins required for LilyGo T-CAN485 board
output:
  - platform: gpio
    id: ENABLE_PIN # Enable the chip
    pin:
      number: GPIO19
      inverted: true
  - platform: gpio
    id: SE_PIN # Enable autodirection
    pin:
      number: GPIO17
      inverted: true
  - platform: gpio
    id: ENABLE_5V_PIN # Enable 5V pin for RS485 chip
    pin:
      number: GPIO16
      inverted: true

# Configure uart that will be used
uart:
  rx_pin: GPIO21
  tx_pin: GPIO22
  baud_rate: 9600

# Configure NibeGW
nibegw:
  udp:
    # The target address(s) to send data to. May be a multicast address.
    target:
      - ip: 192.168.2.5
        port: 10090

    # List of source address to accept data from, may be empty for no filter
    source:
      - 192.168.2.5

  acknowledge:
    - MODBUS40
    
  # Constant replies to certain requests cabe made
  constants:
    - address: MODBUS40
      token: ACCESSORY
      data: [
            0x0A, # MODBUS version low
            0x00, # MODBUS version high
            0x01, # MODBUS address?
      ]

# Some helper functions to restart ESPHome from HA
button:
- platform: restart
  name: Nibegw Restart
- platform: safe_mode
  name: Nibegw Safe Mode Boot
```

elupus, 2023-07-30:
Could you have 15 minutes renew intervall of dhcp? It could theoretically interfer with the execution.

yannik1015, 2023-07-30:
Sadly no, I now set the device to a fixed IP address but the alarm mode continues with the same interval.

elupus, 2023-07-30:
Its your network access point far away? Does it look to have logged anything at those intervals?

yannik1015, 2023-07-30:
The connection to it isn't the best but it works as I can transmit the values I want and haven't seen any issues with it apart from the Modbus Error every 15 minutes. 
Also I am using an Home Assistant Automation to automatically reset the alarm as soon as it occurs, therefor I don't think the connection is the problem.

elupus, 2023-07-30:
We have soon semi lockups when its connecting to networks. So i suggest you try to get some accesspoint closer to it. Perhaps some range extender.

closed 2023-07-31 by yannik1015

elupus, 2023-07-31:
Nice to hear it was resolved. Its annoying that it still causes issues. Though :(