elupus/esphome-nibe #37: Possible to see or set holiday dates and settings?

rasmusbe, 2023-09-26:
Hi,

It seems like I can see if the holiday mode is activated, but not the dates nor settings

From Home Assistant
[image]

Wanted settings, found in the Nibe Uplink App
[image]

elupus, 2023-10-18:
On the physical pump? What menu?

albatorsk, 2023-10-19:
I've set my holiday schedule in the heatpump to a 25 year span from now. Thay way, I can use the Holiday activation toggle in Home Assistant without caring about the dates. However, once holiday mode has been activated, the toggle becomes unavailable and I can only deactivate it again from the physical panel on the heat pump.

rasmusbe, 2023-10-19:
Right now the integration supports the activation setting (48043), but not the options for start/end date (48044 / 48045), water comfort (48047), ventilation (48048) or temperature (48051).

<details>
<summary>Parameter data from Uplink</summary>

```json
[
  {
    "parameterId": 48043,
    "name": "48043",
    "title": "Holiday",
    "designation": "",
    "unit": "",
    "displayValue": "active",
    "rawValue": 10
  },
  {
    "parameterId": 48044,
    "name": "48044",
    "title": "Holiday",
    "designation": "",
    "unit": "",
    "displayValue": "2023-08-24",
    "rawValue": 6079
  },
  {
    "parameterId": 48045,
    "name": "48045",
    "title": "Holiday",
    "designation": "",
    "unit": "",
    "displayValue": "2023-08-30",
    "rawValue": 6085
  },
  {
    "parameterId": 48047,
    "name": "48047",
    "title": "Holiday",
    "designation": "",
    "unit": "",
    "displayValue": "economy",
    "rawValue": 0
  },
  {
    "parameterId": 48048,
    "name": "48048",
    "title": "Holiday",
    "designation": "",
    "unit": "",
    "displayValue": "speed 2 (30%)",
    "rawValue": 2
  },
  {
    "parameterId": 48051,
    "name": "48051",
    "title": "Holiday",
    "designation": "",
    "unit": "°C",
    "displayValue": "18.0°C",
    "rawValue": 180
  }
]
```

</details>

<details>
<summary>Paramerter data captured from the app menu</summary>

```json
[{
		"current": true,
		"textOn": "aktiv",
		"textOff": "inaktiv",
		"valueOn": 10,
		"valueOff": 0,
		"disableIds": null,
		"parameterId": 48043,
		"title": "aktiverad",
		"dependId": null,
		"dependValues": null,
		"invalidValues": null,
		"isQueued": false,
		"type": "CHECKBOX"
	}, {
		"dateLower": "2023-08-24",
		"dateHigher": "2023-08-30",
		"parameterIdLower": 48044,
		"parameterIdHigher": 48045,
		"titleLower": "startdatum",
		"titleHigher": "slutdatum",
		"isQueuedLower": false,
		"isQueuedHigher": false,
		"type": "DATE_RANGE"
	}, {
		"min": 5.0,
		"max": 30.0,
		"def": 20.0,
		"step": 0.5,
		"current": 18.0,
		"unit": "°C",
		"parameterId": 48051,
		"title": "önskad rumstemperatur",
		"dependId": null,
		"dependValues": null,
		"invalidValues": null,
		"isQueued": false,
		"type": "SLIDER"
	}, {
		"current": 0,
		"choices": [{
			"value": -1,
			"text": "från"
		}, {
			"value": 0,
			"text": "ekonomi"
		}, {
			"value": 1,
			"text": "normal"
		}, {
			"value": 2,
			"text": "lyx"
		}],
		"parameterId": 48047,
		"title": "varmvattenkomfort",
		"dependId": null,
		"dependValues": null,
		"invalidValues": null,
		"isQueued": false,
		"type": "ENUM"
	}, {
		"current": 2,
		"choices": [{
			"value": 0,
			"text": "normal (63%)"
		}, {
			"value": 1,
			"text": "hastighet 1 (0%)"
		}, {
			"value": 2,
			"text": "hastighet 2 (30%)"
		}, {
			"value": 3,
			"text": "hastighet 3 (80%)"
		}, {
			"value": 4,
			"text": "hastighet 4 (100%)"
		}],
		"parameterId": 48048,
		"title": "ventilation",
		"dependId": null,
		"dependValues": null,
		"invalidValues": null,
		"isQueued": false,
		"type": "ENUM"
	}]
```

</details>

<details>
<summary>Screenshot from Uplink app for reference</summary>

[image]

</details>

elupus, 2023-10-19:
Nice with the raw data from uplink! We can amend the known databases here: https://github.com/yozik04/nibe

rasmusbe, 2023-10-19:
Not sure how to map/parse the dates thou, seems to be new type of input/output and my python skills are very limited

elupus, 2023-10-19:
Seems to be days since Monday, 1 January 2007

elupus, 2023-10-19:
We would probably need to set some special unit for this which we handle specially

spitfire, 2023-10-19:
> Seems to be days since Monday, 1 January 2007

Nibe time, like the Unix time :D

rasmusbe, 2023-10-20:
Nice, I managed to include them in a local version of the nibe library, and when the date PR is done that should be even better.

But I have a problem with the temperature setting that I need some help with. With the setting 18 °C I get
"[nibe.nibe.connection.nibegw] Ignoring coil 48051 value b4000000 - failed to decode" in home assistant.

"b4000000" is hex for "180 0 0 0" according to https://conv.darkbyte.ru/, so I guess we need some kind of parsing of that value as well

yozik04, 2023-10-20:
@rasmusbe I think you have added wrong settings for these. I added b4000000 to a test and it has successfully decoded it to 18 degrees.

rasmusbe, 2023-10-27:
@yozik04 had to pause this for some days but I could look at it again now,
this is what I've added to the json
```json
 "48051": {
    "title": "Holiday - Room Temperature",
    "size": "u8",
    "factor": 1,
    "name": "holiday-room-temperature-48051",
    "write": true,
    "unit": "\u00b0C",
    "min": 5.0,
    "max": 30.0,
    "default": 20.0
  },
```

and then I get 
`2023-10-27 08:44:53.590 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring coil 48051 value b4000000 - failed to decode`

yozik04, 2023-10-27:
Temperatures are always signed and 16 bit.

rasmusbe, 2023-10-30:
I've added an PR to add the parameters now
https://github.com/yozik04/nibe/pull/124

Thanks for all the help ❤️ 
Next step should be to add support for date type in the home assistant integration.

Closing this issue now.

closed 2023-10-30 by rasmusbe