elupus/esphome-nibe #43: Broken Install after last commit

tudstudent, 2023-10-20:
After the last commit I can validate the config, but I am unable to install:

```
 udp:
    # The target address(s) to send data to. May also be multicast addresses.
    target:
      - ip: 192.168.0.20
        port: 9999

    # List of source address to accept read/write from, may be empty for no filter, but
    # this is not recommended.
    source:
      - 192.168.0.20
```

```
INFO ESPHome 2023.9.3
INFO Reading configuration config/esp32-mqtt-nibe.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esp32-mqtt-nibe (board: esp32-c3-devkitm-1; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
Dependency Graph
|-- WiFi @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Update @ 2.0.0
|-- noise-c @ 0.1.4
|-- ESP32 Async UDP @ 2.0.0
Compiling .pioenvs\esp32-mqtt-nibe\src\esphome\components\nibegw\NibeGwComponent.cpp.o
Compiling .pioenvs\esp32-mqtt-nibe\src\esphome\core\color.cpp.o
Compiling .pioenvs\esp32-mqtt-nibe\src\esphome\core\component.cpp.o
Compiling .pioenvs\esp32-mqtt-nibe\src\esphome\core\component_iterator.cpp.o
Compiling .pioenvs\esp32-mqtt-nibe\src\esphome\core\controller.cpp.o
Compiling .pioenvs\esp32-mqtt-nibe\src\esphome\core\entity_base.cpp.o
Compiling .pioenvs\esp32-mqtt-nibe\src\esphome\core\helpers.cpp.o
Compiling .pioenvs\esp32-mqtt-nibe\src\esphome\core\log.cpp.o
Compiling .pioenvs\esp32-mqtt-nibe\src\esphome\core\scheduler.cpp.o
src/esphome/components/nibegw/NibeGwComponent.cpp: In member function 'void NibeGwComponent::callback_msg_received(const byte*, int)':
src/esphome/components/nibegw/NibeGwComponent.cpp:33:59: error: no matching function for call to 'ip_addr::ip_addr(std::__tuple_element_t<0, std::tuple<esphome::network::IPAddress, int> >&)'
         ip_addr_t address = (ip_addr_t)std::get<0>(*target);
                                                           ^
In file included from C:/Users/filedir/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/include/lwip/lwip/src/include/lwip/sockets.h:46,
                 from C:/Users/filedir/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/include/lwip/port/esp32/include/sys/socket.h:33,
                 from src/esphome/components/socket/headers.h:157,
                 from src/esphome/components/socket/socket.h:6,
                 from src/esphome\components\api\api_frame_helper.h:14,
                 from src/esphome\components\api\api_connection.h:3,
                 from src/esphome.h:3,
                 from src/esphome/components/nibegw/NibeGwComponent.h:9,
                 from src/esphome/components/nibegw/NibeGwComponent.cpp:1:
C:/Users/filedir/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/include/lwip/lwip/src/include/lwip/ip_addr.h:69:16: note: candidate: 'ip_addr::ip_addr()'
 typedef struct ip_addr {
                ^~~~~~~
C:/Users/filedir/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/include/lwip/lwip/src/include/lwip/ip_addr.h:69:16: note:   candidate expects 0 arguments, 1 provided
C:/Users/filedir/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/include/lwip/lwip/src/include/lwip/ip_addr.h:69:16: note: candidate: 'constexpr ip_addr::ip_addr(const ip_addr&)'
C:/Users/filedir/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/include/lwip/lwip/src/include/lwip/ip_addr.h:69:16: note:   no known conversion for argument 1 from 'std::__tuple_element_t<0, std::tuple<esphome::network::IPAddress, int> >' {aka 'esphome::network::IPAddress'} to 'const ip_addr&'
C:/Users/filedir/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/include/lwip/lwip/src/include/lwip/ip_addr.h:69:16: note: candidate: 'constexpr ip_addr::ip_addr(ip_addr&&)'
C:/Users/filedir/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32c3/include/lwip/lwip/src/include/lwip/ip_addr.h:69:16: note:   no known conversion for argument 1 from 'std::__tuple_element_t<0, std::tuple<esphome::network::IPAddress, int> >' {aka 'esphome::network::IPAddress'} to 'ip_addr&&'
src/esphome/components/nibegw/NibeGwComponent.cpp: In member function 'void NibeGwComponent::token_request_cache(AsyncUDPPacket&, byte, byte)':
src/esphome/components/nibegw/NibeGwComponent.cpp:60:41: error: conversion from 'IPAddress' to non-scalar type 'esphome::network::IPAddress' requested
     network::IPAddress ip = udp.remoteIP();
                             ~~~~~~~~~~~~^~
Compiling .pioenvs\esp32-mqtt-nibe\src\esphome\core\string_ref.cpp.o
Compiling .pioenvs\esp32-mqtt-nibe\src\esphome\core\time.cpp.o
Compiling .pioenvs\esp32-mqtt-nibe\src\esphome\core\util.cpp.o
Compiling .pioenvs\esp32-mqtt-nibe\src\main.cpp.o
*** [.pioenvs\esp32-mqtt-nibe\src\esphome\components\nibegw\NibeGwComponent.cpp.o] Error 1
Building .pioenvs\esp32-mqtt-nibe\bootloader.bin
Creating esp32c3 image...
Successfully created esp32c3 image.
========================= [FAILED] Took 11.47 seconds =========================
```

elupus, 2023-10-20:
Ok. Something differs for c3 esp. Will investigate.

tudstudent, 2023-10-20:
My bad, I just updated to ESPHome  esphome-2023.10.1 (I was on 2023.9.3), now it compiles and updates the ESP.
I do not know if this is coincidence or meant.
However if I need to assist in a test, let me know.

elupus, 2023-10-20:
It was meant. Good that it works.

closed 2023-10-20 by elupus