elupus/esphome-nibe #46: RS485 & GND Issues -> Tip for readme

tudstudent, 2023-10-30:
Hi @elupus 

Recently I have been playing around with ESP and RS485 (MAX485). After a lot of researching I got it working well, but it might be useful for other users to mention:

- If you want to use the **12V supply from the Nibe HP** (terminal 9 and 12), you need to use an **ISOLATED RS485** module

It is especially helpful to use the 12V from the HP, because in case of outage, this will restart the ESP in parallel, ensuring power to the unit and therefore no error. Additionally you can in this way place the PCB's inside the HP without additional adapter or power supply.

If you do still use the GND from the Nibe and do not use isolated RS485 you will see only A OR B channel will work and it uses the GND for the other connection (ground loop).

You need a LM2596 DC-DC converter to convert from 12V to 5V (or likewise).

elupus, 2023-10-30:
Could you expand a bit on this. Im not following why you need it to be isolated. Is the gnd for rs485 and the gnd for 12v on the pump not common?

tudstudent, 2023-10-30:
On the HP, we have 4 terminals (GND, 12V and A&B).
When I use the 12V with the GND to power the ESP and the RS485 I see that the HP goes immediately in error.
When I disconnect the RX (A or B dunno), I see the RS485 connection works.... Strange.
(HP then stays working)

When connecting the ESP and RS485 board to a separate power supply and only connect A&B form HP, then the HP also functions fine.
If connecting the GND additionally to the A&B channel (to create common GND), it will also immediately go in error.

So only works with:

1. Separate PS and A&B connected (no GND connected)
2. Presumably PS from HP with an isolated RS485 board

However not reading a lot on the schematics of other users I presume all separate PS
Hope this is clearer

tudstudent, 2023-10-30:
Currenlty I sue a seperate 5V PSU.
However I want to use the 12V from the HP. This can be done by using a LM2596 (for example) which adjust the 12V to 5V.
Meaning you also need to use the GND (and that is then disabling the RS485 wokrings -> errors)

elupus, 2023-10-30:
Right. Just wanted to make sure. I dont see why it would need to be isolated still unless something binds a or b to gnd.

tudstudent, 2023-10-30:
This is the board used:
https://protosupplies.com/product/max485-ttl-to-rs-485-interface-module/
(Bought on ali, but same product)

ESP= ESP32 => ETH01-EVO
Power converter => LM2596 

Cable between HP and MAX485 = unshielded CAT5
In breadboard, jumper wired to A & B pins (not using the screw terminals.
As said, if I connect GND of both devices (HP & Max), will result in not working.

elupus, 2023-10-30:
I dont really know what is the cause. But would be very good if you can figure it out. Could you have some issue with the power regulator causing it to offset ground weirdly?

I suppose the gnd pin of the HP might be a different ground plane from what the rs485 com is using, but seem really weird.

bgarderhagen, 2023-11-09:
Hi.

I have been struggeling with Modbus comms error alarms a few times each day. Tried a lot of different stuff. Made an automation that reset the alarm as a workaround.

I used NibeGW on M5 Atom with RS-485 powered from terminal 9 and 12.

[image]

Yesterday i replaced it with Lilly CAN-RS-485 connected as the example. This has been working without any issues so far.
Might be related to the GND potential issue you mention.

[image]

DaRkMaN-FX, 2023-11-18:
@bgarderhagen 
How did you get the code on the Lilly-GO ?
I have always errors during compiling

Can you give an example of your code you are using?

bgarderhagen, 2023-11-18:
The code is deployed to the Lilly-GO by ESPHome integration in Home Assistant. The yaml file refers to git where the code is located.

tudstudent, 2023-11-23:
@elupus I got some time yesterday and investigated the Lillygo design:
https://github.com/Xinyuan-LilyGO/T-CAN485/blob/main/doc/schematic.pdf

It seems the LillyGo uses an 13487E:
https://www.analog.com/media/en/technical-documentation/data-sheets/MAX13487E-MAX13488E.pdf

Compared to the MAX485 which most use:
https://www.analog.com/media/en/technical-documentation/data-sheets/MAX1487-MAX491.pdf

Another difference (convenience) is that the LillyGo has AutoDirection (but not relevant to this post).

So I suspect you do not have a problem with a common ground, because your RS485 signals are isolated.
If I can get my hands on a 13487 type device I will test my findings and report back.

Disclaimer: I am not an electronics guy from profession, so if someone stops by and thinks this is wrong, please feel free to explain so we can all learn ;-)

tudstudent, 2023-12-15:
Update: Yesterday I received the 13487 breakout. Today installed it in place and connected it to the power of the HP.
Worked directly without any problems.

Might be worth mentioning that IF you want to use the PSU of the Nibe you need a 485 connection, either with a isolation circuit added or by using a chip like the 13487 which has this functionality embedded.
With this I conclude this search and hope other might be helped with this info as well.

DenisMir, 2024-04-14:
Is there a way to make the M5 Stack one work correctly?

elupus, 2025-05-28:
I'm going to close this. One final thing we have noticed. If you use the lilygo TCAN485, you may need to ground both sides of the card to the heatpump. The lilygo is isolated, but the ground on the rs485 side may float too far away from the heatpumps signal levels and thus break the connection.

closed 2025-05-28 by elupus