elupus/esphome-nibe #47: RMU40 Sensor

delabrava, 2023-10-30:
Great work, just one thing I miss, and that's the possibility to emulate a room sensor. Reviewing how that is done in the **_nibepi_** project I see it's done by emulate a **RMU40** where it writes the room temperature to register 10060. 

How can we access same register using this integration(**_esphome-nibe_**), is the limitation of registers maybe within the scope of the  **_nibegw_** project?

{"register":"10060","factor":10,"size":"s16","mode":"R/W","titel":"RMU Temp S1","info":"","unit":"°C","min":"0","max":"0"},

elupus, 2023-10-30:
Its within scope. Just not figured our a good API yet. I might cheat and fix it with home assistant sensors.

delabrava, 2023-10-30:
Thanks for you reply. 

Not sure what you mean by cheating, 😊  maybe it's the same think as I'm doing already?
I've create a template sensor to calculate the heatcurve offset and updating it using an automation. 

I'm closing this, was mostly curios if there where some plans of doing implementing RMU40 registers or not.

egisz, 2023-11-06:
> calculate the heatcurve offset and updating it using an automation.
Hi, I'm new to esphome-nibe and I would like to automate based on 3rd party room temperature sensor (Shelly H&H Plus) as I don't have installed Nibe room sensor...  Is your case same?
Could you share sensor template and automation code?

Or even better, it would be nice to fake RMU40 and send 3rd party sensors temperature pretending as RMU40. This way maybe we don't need  HA automation, but configure in Nibe itself. As I understand in README, full config last lines does that, just constant temperature, right?
```
    # Constant fixed temperature to avoid pump going into alarm.
    - address: RMU40_S4
      token: RMU_WRITE
      data: [
            0x06, # Temperature
            0x14, # degrees low
            0x00, # degrees high
      ]
```

elupus, 2023-11-06:
Yup. I am split. On one hand i want the gw unlt to be as simple as possible. On the other hand pretending to be a RMU within standard esphome infra would also be cool.

egisz, 2023-11-06:
I replaced RMU40_S4 to RMU40_S1, enabled in Nibe 5.2.4 menu and it now shows 1C indoor temperature.
I cannot understand why it shows such small number, but in general it proves that it works.
Could you explain how to encode right temperature?

egisz, 2023-11-08:
I managed to fake RMU40 sensor, created [PR#52](https://github.com/elupus/esphome-nibe/pull/52).
Feedback is very welcome.