elupus/esphome-nibe #48: Conversion of IPAddress to non-scalar type

Pivk77, 2023-10-30:
I get the following error with the provided example and changed for my use case

[image]

elupus, 2023-10-30:
Please upgrade the component

closed 2023-10-30 by elupus

Pivk77, 2023-10-30:
And how do i do that? i guess it pulls the code directly from github?

elupus, 2023-10-30:
It does, but you may have to clear cache in esphome in some way.

Pivk77, 2023-10-30:
I deleted everything and redownloaded via homeassistant and manualy and still the same issue

Pivk77, 2023-10-30:
I'm using esphome dev and yes it is updated ... could the issue be the dev esphome addon?

elupus, 2023-10-30:
Maybe. Yes. But if it is then i need to figure out why dev doesnt work.

Pivk77, 2023-11-01:
[image: image.png][image: image.png][image: image.png]

Hi,
since i have your attention ... i'm still struggling to get the thing
working ... i have a VVM310 HP, and i cant find the enable MODBUS40 in
settings and i cant get it connected. I guess the modbus connection is ok.
I also tried different ports if some were blocked

Pivk77, 2023-11-01:
also i get this error ... even if i have the devices on the same network,
but since i have homeassistant on 192.168.20.3 and esp on 192.168.20.195 it
should be fine

Pivk77, 2023-11-01:
[image: image.png]