elupus/esphome-nibe #56: Step-by-step configuration

Gigatrappeur, 2023-11-20:
Hello,

I'm sorry, this not a real issue but is it possible to add a step-by-step for use lilygo TTGO T-CAN485 ESP32 CAN RS-485 ?
I'm experimented developer but I'm not confortable with esp or arduino...

I saw the photo for the electrical connections, this very helpful.
But I need a procedure for how to plug in the card, compile the code and download it to the card?
Many thanks in advance

I'd really like to use this board rather than the official Nibe modbus module which costs ~700€

Best regards

bgarderhagen, 2023-11-21:
Hi.

Here are some steps from how i remember it quickly:

- Add ESPHome integration in Home Assistant
- Go to the ESPHome dashboard added on left side
- Click new device
- "Open ESPHome"
- Connect your LillyGo by USB cable to the computer running the browser
- Connect and deploy to new device
- Edit the YAML file. Look at the example file on git
- Install YAML file to ESP

This will deploy nibegw to the LillyGo.

Then add the Nibe integration and connect it to the ip of the nibegw.

Hope this helps you along.

Can make a video or something later if required

//btg

waelzindani, 2023-12-28:
This is amazing. Just got my card and I will install it in the weekend. If there is any video pls paste the link.
Have a nice evening!

vesar-github, 2024-01-20:
Hi!

I just received my Lilygo board today. As a fresh Home Assistant user and absolute first timer for ESP32, I found this step by step. I will record here what I did and how far I got in a bit more detail for others to follow. Maybe someone can help me onward or comment about best practices for the next one to install their setup. I will update this with any corrections you may point out.

**As said, I am a total newbie. My procedure may be totally wrong.**

Thanks to  @bgarderhagen for the step by step to start with. @Gigatrappeur, @waelzindani, did your installation go ok?

My setup: Home assistant running on Raspberry Pi 4b with HAOS + another desktop PC as UI. No ESPHome anything installed in the house beforehand.

1. In UI PC, in Home assistant, go to Settings -> Add-ons and search for ESPHome
[image]

2. Click ESPHome
[image]

3. Click Install
[image]

4. After Installation, Click START and then OPEN WEB UI
[image]
[image]

5. In WEB UI, click + NEW DEVICE
[image]

6. UI computer warns about not browsing the dashboard with a HTTPS-connection, but that You can connect the device to the computer running ESPHome dashboard. That will be the Raspberry with HAOS. No worries if You have access to that computer. So continue.
[image]

8. Give a name to the configuration and credentials to your Wifi, where the LILYGO will reside.
[image]

9. Configuration is created. Press INSTALL.
[image]

10. Connect the LILYGO board to the computer running ESPHome dashboard with USB. In this case, the Home Assistant server.
11. Select "Plug into the computer running ESPHome dashboard"
[image]

12. The device is found, select device.
[image]

13. Installation starts and takes a couple of minutes. You can see log running as installation proceeds.
[image]

14. After successful installation, you can see your LILYGO device in your network
15. You should assign a static IP for your LILYGO in your network router.
16. In Home Assistant go to Settings -> Add-ons -> ESPHome -> OPEN WEB UI, and you should see the device you just created being ONLINE
[image]

17. Go to settings -> Devices and services. You should now see ESPHome as newly found. Add that. You can see one device with no information.
18. In Home Assistant go to Settings -> Add-ons -> ESPHome -> OPEN WEB UI
19. Click SECRETS
[image]

20. Add a row for OTA_PASSWORD and set your own password
[image]

21. Change default generated names WIFI_SSID and WIFI_PASSWORD to lowercase wifi_ssid and wifi_password to comply with the nibegw code example - or modify the code example later to use lowercase, if you already use the names elsewhere
> 21b. Just a newbie question, are these case sensitive? Was this item necessary at all?

22. In your new device, click EDIT to edit the code. In my case, the default code is as follows:

```
esphome:
  name: nibe-f1255
  friendly_name: Nibe_F1255

esp32:
  board: esp32dev
[... 117 lines cut ...]

# Some helper functions to restart ESPHome from HA
button:
- platform: restart
  name: Nibegw Restart
- platform: safe_mode
  name: Nibegw Safe Mode Boot 

```
26. Click SAVE
27. Click INSTALL and follow the procedure as before in items 10...
[image]

28. Disconnect your LILYGO from Home Assistant computer and supply it power via USB. Check that it becomes online wirelessly and taht you can see it ONLINE in Home Assistant by going to Settings -> Add-ons -> ESPHome -> OPEN WEB UI

After an amount of fiddling, I got the following to show in my Home Assistant
[image]

So I can only see 1 device and 3 entities. Maybe this is correct, maybe not.

This is as far as I went now. it's -27C outside and saturday, so today is not the best day to make experimental connectios to our only heat source and burn its brain :) I'll go on with installation on monday and maybe even get some feedback from others before that so I can correct my mistakes :)

To be continued...

Vesa

waelzindani, 2024-01-20:
Hello
This is great. I started and finished the first part which is ESP but the weather has been bloody cold here and I havent had the time to finish. You are a much better newbie than I am haha.

Gigatrappeur, 2024-01-23:
Hello guys,  
Yes, I've successfully installed lyligo card.
Some initial difficulties with connecting the card to ESPHome, but in the end it seems to work.
  
However, some values are missing, such as the outside temperature.
I have a probe connected to the nibe and the nibe displays this temperature on its screen. But I don't know how to retrieve it in HA.

elupus, 2024-01-23:
Enable the sensor

elupus, 2024-01-23:
Whatever you do, dont enable all sensors. That will cause really low update rate. The more you enable the slower it gets. Some senors can also break the update causing unavailable data.

Start with only the ones in your logset. Then enable more as you need them.

vesar-github, 2024-01-23:
I had 7 sensors enabled, 3 of them were somehow enabled by "default" or automatically when I installed the integration. Now I dropped it to three, only the phase currents left. Still a lot of unavailability, no visible change in how the history graph is drawn.

Makes me wonder, did my heat pump accept the LOG.SET. Is there an easy way to verify this? Then again, even if not, i guess it should be able to regularly update three values.

tobbensoft, 2024-01-23:
Maybe a stupid question, but where do I find ModbusManager,  using esphome-nibe with success but a better update speed on some values would be great :)

vesar-github, 2024-01-23:
Nibe MODBUS 40 manual says "ModbusManager can be downloaded via www.nibe.se/kommunikation". That takes you to a page, where you can find menu PROFFSHJÄLP (Professional help) -> KOMMUNIKATION -> NIBE Modbus. Direct link [here](https://professional.nibe.eu/sv/proffshjalp/kommunikation/nibe-modbus).

I had trouble finding similar in Nibe Finnish home pages.

vesar-github, 2024-01-23:
Nothing fishy in ESPHome DEBUG logs as far as I understand.

Setting logging to VERY_VERBOSE was too much for the ESP to handle, the heat pump went to alarm state. Nevertheless, I collected some amount of log [logs_nibe-f1255_logs_VV_.txt](https://github.com/elupus/esphome-nibe/files/14029423/logs_nibe-f1255_logs_VV_.txt). I just can't say, is there something I should worry about, and is that just because verbose logging overloaded the ESP.

There are several times entries like:
```
[21:52:14][W][component:214]: Component nibegw took a long time for an operation (0.05 s).
[21:52:14][W][component:215]: Components should block for at most 20-30ms.
```
but i didn't see these when logging was VERBODE or DEBUG. Is that because these are very verbose only, or did this only happen when overloading the ESP?

The part, which I understand more of, seems to be ok. Shielded cable of 30 cm length between the Nibe and LILYGO should not produce errors to communication.

Then I set Nibe integration to debug logging, and this kind on rows were present:

`2024-01-23 22:31:56.392 DEBUG (MainThread) [homeassistant.components.nibe_heatpump] Some coils failed to update, and may be unsupported: Failed to read some or all coils (Failed decoding response for eb100-be1-current-40083: Failed to decode eb100-be1-current-40083 coil from raw: 6a00, exception: Invalid raw data size: given 2, expected at least 4)`

A bit longer sample:
```
2024-01-23 22:29:41.646 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring coil 40080 - coil definition not found
2024-01-23 22:29:43.633 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring coil 40083 value 6c00 - failed to decode
2024-01-23 22:29:43.633 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring coil 40084 - coil definition not found
2024-01-23 22:29:43.634 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring coil 40081 value 3c00 - failed to decode
2024-01-23 22:29:43.634 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring coil 40082 - coil definition not found
2024-01-23 22:29:43.635 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring coil 40079 value 4700 - failed to decode
2024-01-23 22:29:43.635 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring coil 40080 - coil definition not found
2024-01-23 22:29:46.628 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring coil 40083 value 6c00 - failed to decode
2024-01-23 22:29:46.628 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring coil 40084 - coil definition not found
2024-01-23 22:29:46.629 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring coil 40081 value 3b00 - failed to decode
2024-01-23 22:29:46.629 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring coil 40082 - coil definition not found
2024-01-23 22:29:46.630 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring coil 40079 value 4500 - failed to decode
2024-01-23 22:29:46.630 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring coil 40080 - coil definition not found
2024-01-23 22:29:47.032 DEBUG (MainThread) [homeassistant.components.nibe_heatpump] Skipping seeded coil: 40079
2024-01-23 22:29:48.661 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring coil 40083 value 6c00 - failed to decode
2024-01-23 22:29:48.661 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring coil 40084 - coil definition not found
2024-01-23 22:29:48.662 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring coil 40081 value 3b00 - failed to decode
2024-01-23 22:29:48.662 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring coil 40082 - coil definition not found
2024-01-23 22:29:48.662 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring coil 40079 value 4500 - failed to decode
2024-01-23 22:29:48.663 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring coil 40080 - coil definition not found
2024-01-23 22:29:50.433 DEBUG (MainThread) [homeassistant.components.nibe_heatpump] Some coils failed to update, and may be unsupported: Failed to read some or all coils (Failed decoding response for eb100-be1-current-40083: Failed to decode eb100-be1-current-40083 coil from raw: 6c00, exception: Invalid raw data size: given 2, expected at least 4)
2024-01-23 22:29:50.434 DEBUG (MainThread) [homeassistant.components.nibe_heatpump] Finished fetching Nibe Heat Pump data in 3.402 seconds (success: True)
2024-01-23 22:29:50.666 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring coil 40083 value 6900 - failed to decode
2024-01-23 22:29:50.667 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring coil 40084 - coil definition not found
2024-01-23 22:29:50.667 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring coil 40081 value 3f00 - failed to decode
```

elupus, 2024-01-23:
Interesting. Seems we are getting a lot of data we dont underatand. Could you remove the log.set file again? Looks almost like it might be sending data in a weird format.

vesar-github, 2024-01-23:
I actually did make another LOG.SET with only those three currents to try, did not help.

Now I left logging not activated and even selected "Ignore LOG.SET" in the heat pump, what ever that did. It's difficult to say with the heat pump UI, because you lose the menu once you remove the USB stick. Was that what you mean by removing the LOG.SET? At least the data did not get any better, still unavailable a lot.

Looking at Nibe Debug log a bit wider, some checksum rows also stand out from "background noise". Quite a few of those, and then the checksum errors just suddenly stop appearing.

```
2024-01-23 21:42:50.453 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring packet from ('192.168.1.235', 9999) due to checksum error: 5c00206900df
2024-01-23 21:42:51.286 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring packet from ('192.168.1.235', 9999) due to checksum error: 5c00e6d02064c45c00025502087d205c00027e92155c0002900092065c00206b004b5c00206b
...
2024-01-23 21:47:39.859 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring packet from ('192.168.1.235', 9999) due to checksum error: 5c0020685085a84c006fa94f0080a8c7006d9f0000a3a93c00fda735fccc6721939c6a00949c0000919c4000929c00008f9c4500909c0000ffff0000ffff0000ffff0000ffff0000ffff0000ffff0000ffff0000ef5c
...
2024-01-23 21:48:14.914 DEBUG (MainThread) [homeassistant.components.nibe_heatpump] Skipping seeded coil: 40083
2024-01-23 21:48:16.635 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring packet from ('192.168.1.235', 9999) due to checksum error: 5c0020685085a84c006fa94f0080a8c8006d9f0000a3a93c00fda735fc449c0b00939c6c00949c0000919c7d48c606008f9c4600909c0000ffff0000ffff0000ffff0000ffff0000ffff0000ffff0000ffff0000985c
2024-01-23 21:48:17.780 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring packet from ('192.168.1.235', 9999) due to checksum error: 5c0020685085a84c006fa94f0080a8c8006d9f0000a3a93c00fda735fc449c0b00939c6c98190600919c3d00929c00008f9c4600909c0000ffff0000ffff0000ffff0000ffff0000ffff0000ffff0000ffff0000985c
2024-01-23 21:48:22.484 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring packet from ('192.168.1.235', 9999) due to checksum error: 5c0020685085a84c006fa94f0080a8c8006d9f0000a3a93c00fda735fc7fb600939c6c00949c0000919c3d00929c00008f9c4700909c0000ffff0000ffff0000ffff0000ffff0000ffff0000ffff0000ffff0000995c
...
2024-01-23 21:48:43.516 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring packet from ('192.168.1.235', 9999) due to checksum error: 5c0002a0321016fc5c00025502087d205c0002900092065c0002a0020064c45c00025502087d205c0002900092065c0002a0020064c45c00
```

Latest log attached. Thank you already for the support, hope to find out something that might help others in the future too.
[home-assistant_nibe_heatpump_2024-01-23T21-28-48.910Z.log](https://github.com/elupus/esphome-nibe/files/14029976/home-assistant_nibe_heatpump_2024-01-23T21-28-48.910Z.log) 

I'll be traveling for work for a couple of days, will get back to this when I'm back home again.

elupus, 2024-01-24:
The checksum issues can occur of the esp is overloaded. If you still have it set to verbose logging, turn that down.

The other warnings looks to be due to "EB100-BE3 Current" values enabled in LOG.SET. They are 32bit and seem to have been split over two parameters.

I don't think you managed to disable the LOG.SET in your last log.

tobbensoft, 2024-01-24:
I tried using LOG.SET and got similar results to you @vesar-github 

I think I managed to disable it by putting an empty LOG.SET file on the usb and doing the process again

vesar-github, 2024-01-25:
Logging was already dropped to VERBOSE, now at DEBUG (little difference between VERBOSE and DEBUG).

OK, I also made an empty LOG.SET file. Seems to be disabled now, although it did nothing to history graph.

In addition to the 32 bit currents, I added some temperatures too that are 16 bit values. And yes, they seem to graph fine! Very low update interval, but one problem at a time. So is it something in my setup that messes up the 32 bit values, or is there something in the way nibegw handles 32 bit values?

[image]

elupus, 2024-01-25:
32bit values in log set is known to be buggy.

32bit values can be broken if your wordswap setting is wrong.

tobbensoft, 2024-01-26:
I have graphed 32-bit degree minutes for some time now,  that works fine..  Is it only some 32-bit values?

Is that the problem we get with LOG.SET values or is that a different thing?

vesar-github, 2024-01-26:
Confirmed. I also added 32 bit degree minutes, and that draws a nice, continuous graph. Phase current values are often unavailable. 

LOG.SET is now empty.

vesar-github, 2024-01-30:
@elupus, any ideas on where this issue (some 32 bit values being constantly ok and some constantly not) might originate? Heat pump, NibeGW, configuration, something else? In other words, is there anything one can do?

Another thing is the update frequency. @tobbensoft, how often do you get updates? My values update about once per minute, no matter if there is one or twenty enabled entities. Any ideas how to chenck, whether the heat pump is slow to answer or Home Assistant is slow to ask? My assumption is that a much faster update frequency should be possible. My assumption may me wrong :)

tobbensoft, 2024-01-30:
@vesar-github I have a lot of entities enabled, so my update frequency is maybe 2mins. But i would say I have no problems with the values as such,  just a little slow compared to other technologies.

It would be nice to be able to prioritize the one important in the LOG.SET and the other ones just comes as we roll through the enabled entities.

Just added the EMK 500 flowmeter to my VVM 500 / F2120-20 so I can measure produced heat in kWh :)

elupus, 2024-01-30:
32bit values from log.set are broken. Otherwsie they should work as long as your wordswap setting is correct on the pump.

vesar-github, 2024-01-30:
@tobbensoft Yeah... My specific interest are just the phase current values (which are now broken) and with high update frequency (which is not the case now) :( This way I would avoid installing a second set of current transformers to my already crammed main electric panel for load balancing.

Well, still investigating. And considering a Shelly pro 3EM...

vesar-github, 2024-02-02:
Thanks a lot @elupus, finally got this confirmed. I made a log.set with just "phase 2 current" in it. And yes, the two other phases made a nice graph, only phase 2 got unavailable values. Is this something wrong with Nibe heat pump, as in would the official modbus accessory produce the same result anyway or it this something that has not been reverse engineered yet?

Experimenting with a couple of 16 bit temperature values, both om them in log.set had about 1...2 second update frequency, while removing one of them lowered its update frequency to about one minute. As a matter of fact, no matter if I have 1 or 20 entities enabled, the update frequency outside log.set seems to be about one minute. By your experience, is such a low refresh rate normal behavior? Reading MODBUS 40 manual, I would expect something faster. @tobbensoft, could you please repeat this test with only one value enabled in HA? Same results?
```
Manual readout is time consuming and only
one value at a time can be read, max timeout
2,1 s. The parameter does not have to be in-
cluded in the LOG.SET file. 
```
[Manual](https://www.nibe.fi/assets/documents/15050/031725-6.pdf)

elupus, 2024-02-02:
Its wrong in our parsing of logset 32bit values.

Refresh rate is by default 1minute. We dont want to overload the pump. There is at the moment no way to tune individual entities refresh rate. You can ask any entity to refresh with an automation (update entity service), but thay will cause a full refresh of all entities.

tobbensoft, 2024-02-03:
@vesar-github I got similar results to you,  LOG.SET works if you avoid 32-bit values and updates are nice and speedy.

Around 1 minute for everything else.

elupus, 2024-02-06:
Fixes for 32:bit values on the way. Will likely miss this release, but for next.. https://github.com/home-assistant/core/pull/109825

vesar-github, 2024-02-06:
@elupus, excellent! If there is a non-endless wish list, being able to update individual entities (non-log.set) with faster interval would also be great. Although in my use case, after this 32 bit fix, it is enough to just put phase currents in log.set and other values can be once a minute.

vesar-github, 2024-02-06:
When starting my journey into Nibe integration, this was my first try at ESPHome, so it required a bit of investigation to actually get the dots connected. This very short Step-by-step issue was the key to success, although still needing some guesswork. That's why I documented my progress in the format above.

Nevertheless, I found this issue by pure luck. I didn't expect to find installation instructions in issues. Frankly, without this, I never would have completed the installation. I would like to have found more detailed instructions in the front page, or a link to them.

@elupus, to help others maybe struggling with the same issue in the future, what would be easier-to-find location for the step-by-step installation? If you can point me to what to do, I could reformat and revise the instructions for easier access to future beginners. Would be my way to contribute, since I'm not skilled enough to contribute code.

realzoulou, 2024-06-13:
Since I updated my F1245 with myUplink firmware, I now try to get NibeGw running in HA, but cannot run through the Config Flow. 
It fails with **Error on write request to pump. Verify your Remote write port or Remote address.** and in HA log **nibe.exceptions.WriteException: Heatpump denied writing alarm-reset-45171**

I have
* LOG.SET with entries 45171, 40004, 44298, 44300
* HA: 2024.6.2 on RPi4 as docker (privileged: true, network_mode: host)
* LilyGo ESP32 T-CAN485 with esphome (followed mainly https://github.com/elupus/esphome-nibe/blob/master/examples/lilygo-t-can485.yaml)
I captured tcpdump on RPi to check the IP traffic between RPi and LilyGo
I captured with a LogicAnalyzer the RS485 communication between LilyGo and Nibe.
I can't find anything fishy in these logs, which would help me find the root cause for the ConfigFlow error.

Maybe @elupus could have a look at below intermixed traces of Wireshark and LogicAnalyzer ?

192.168.178.71 = LilyGo ESP32 T-CAN485
192.168.178.53 = RPi4 HomeAssistant
192.168.178.42 = PC with browser to http://192.168.178.53:8123 (=HomeAssistant web page)

Submit button in the ConfigFlow...
```
Wireshark: 321	20:23:10,668332	192.168.178.42	192.168.178.53	HTTP/JSON	1324	POST /api/config/config_entries/flow/249507256e39f7a9fc443227da27c08f HTTP/1.1 , JSON (application/json)
{"model":"F1245","ip_address":"192.168.178.71","listening_port":9999,"remote_read_port":9999,"remote_write_port":10000}
Wireshark: 323	20:23:10,717763	192.168.178.53	192.168.178.71	UDP	48	9999 → 9999 Len=6 Data: c0690273b068
LogicAnalyzer: sec.msec: 4.71702 bytes:	C0 69 02 73 B0 68 
Wireshark: 333	20:23:11,382506	192.168.178.71	192.168.178.53	UDP	60	9999 → 9999 Len=6 Data: 5c00206b004b
Wireshark: 334	20:23:11,403468	192.168.178.71	192.168.178.53	UDP	60	9999 → 9999 Len=6 Data: 5c0020690049
LogicAnalyzer: sec.msec: 4.734488 bytes: 5C 00 20 68 50 FF FF 00 00 FF FF 00 00 FF FF 00 00 FF FF 00 00 FF FF 00 00 FF FF 00 00 FF FF 00 00 FF FF 00 00 FF FF 00 00 FF FF 00 00 FF FF 00 00 FF FF 00 00 FF FF 00 00 FF FF 00 00 FF FF 00 00 FF FF 00 00 FF FF 00 00 FF FF 00 00 FF FF 00 00 FF FF 00 00 18 
Wireshark: 337	20:23:11,577671	192.168.178.71	192.168.178.53	UDP	128	9999 → 9999 Len=86 Data: 5c00206850ffff0000ffff0000ffff0000ffff0000ffff0000ffff0000ffff0000ffff0000ffff0000ffff0000ffff0000ffff0000ffff0000ffff0000ffff0000ffff0000ffff0000ffff0000ffff0000ffff000018
LogicAnalyzer: sec.msec: 5.673359 bytes: 5C 00 20 6A 06 73 B0 00 00 00 00 8F 
Wireshark: 360	20:23:12,401951	192.168.178.71	192.168.178.53	UDP	60	9999 → 9999 Len=12 Data: 5c00206a0673b0000000008f
Wireshark: 361	20:23:12,405069	192.168.178.53	192.168.178.71	UDP	52	9999 → 10000 Len=10 Data: c06b0673b0000000006e
LogicAnalyzer: sec.msec: 5.739657 bytes: C0 6B 06 73 B0 00 00 00 00 6E 
LogicAnalyzer: sec.msec: 1.672633 bytes: 5C 00 20 6B 00 4B 
Wireshark: 362	20:23:12,428258	192.168.178.71	192.168.178.53	UDP	60	9999 → 9999 Len=6 Data: 5c00206b004b
LogicAnalyzer: sec.msec: 1.695135 bytes: 5C 00 20 69 00 49 
Wireshark: 364	20:23:12,499127	192.168.178.71	192.168.178.53	UDP	60	9999 → 9999 Len=6 Data: 5c0020690049
LogicAnalyzer: sec.msec: 1.722118 bytes: 5C 00 1C 62 1F A9 00 EF 01 00 00 00 00 12 00 00 00 00 00 00 00 0B 14 17 00 01 00 00 02 00 06 0C 00 00 00 00 35 
Wireshark: 371	20:23:12,603037	192.168.178.71	192.168.178.53	UDP	79	9999 → 9999 Len=37 Data: 5c001c621fa900ef010000000012000000000000000b1417000100000200060c0000000035
Wireshark: 388	20:23:13,421695	192.168.178.71	192.168.178.53	UDP	60	9999 → 9999 Len=7 Data: 5c00206c01004d
Wireshark: 389	20:23:13,421695	192.168.178.71	192.168.178.53	UDP	60	9999 → 9999 Len=6 Data: 5c0020690049
```
... error visible in HomeAssistant :-(

waelzindani, 2024-11-17:
Hello!
It worked well for me. It was a bit of a hassle when I didnt reserve the IP adress for the Lilygo but it was solved. Really thank you for the detailed instructions and thanks to everyone involved.

next step is to fiddle deeper and make use of this great integration!
Have a nice Sunday everyone!

krenlund, 2025-03-16:
Thank you @vesar-github! Great instructions. 🙏

ostsee-segler, 2025-08-20:
Hi,
new here and tried to follow the detailed instructions above. I have created the yaml file but when I press Install at step 27 I receive the error:
In file included from src/esphome/components/nibegw/NibeGw.cpp:19:
src/esphome/components/nibegw/NibeGw.h:39:10: fatal error: Arduino.h: No such file or directory

Any idea? As this comes from the external source I have no idea how to fix this.

Thanks in advance
Thomas

elupus, 2025-08-20:
I think you are using a bad example. It should retain the code for which platform to use:

```
esp32:
  board: esp32dev
  framework:
    type: arduino
```

I think you aught to be able to use the template here: https://github.com/elupus/esphome-nibe/blob/master/examples/lilygo-t-can485.yaml.

If anybody here is knowledgeable on esphome templating, would be nice to see our examples be more parametric. So we don't have to maintain so many variants.

ostsee-segler, 2025-08-20:
Perfect, thanks a lot, issue solved!

vesar-github, 2025-08-21:
@ostsee-segler , you succeeded in your installation?

I know that HA frontend has changed a lot during the last year and dialogs are now different in many places. Was the experience still doable with the step-by-step? Just wondering, whether some steps should be re-written?

ostsee-segler, 2025-08-22:
Hi Vesar,
it works pretty well with the current version. The one hint I received above was the only blocker I noticed. Some steps are easier now, i.e. looking at logs from the ESPHome device does not require ssh anymore, you can access it directly from the HA UI.

dolnet, 2025-09-29:
So, step by step, I've gone through preparing the LilyGo ESP32 T-CAN485, and it's time to connect it to Nibe.
I have an SMO20 and an F2040, and communication between these units is via RS485. My question is, should I connect the LilyGo ESP32 T-CAN485 to this communication?
I'd rather ask beforehand than afterhand.

Hami5h, 2026-09-08:
Hi @elupus,

I'm hoping you can help with a register issue on an NIBE F1245-8.

I'm using your `esphome-nibe` NibeGW component with an M5Stack ATOM Lite connected to the F1245 via AA3-X4, and Home Assistant is successfully communicating with the heat pump.

However, I'm having problems with the heat-produced registers. In Home Assistant I have, for example:

`sensor.heat_meter_heat_cpr_and_add_total_system_42439`

currently reporting approximately 1328.80 kWh.

I also have:

`sensor.heat_meter_hw_cpr_and_add_ep14_44298`

reporting approximately 510.40 kWh.

The values look plausible, but they have apparently not changed since the initial integration despite the heat pump continuing to operate.

This makes me suspect that either the wrong registers are being used, the register configuration in the NIBE MODBUS 40 interface is incorrect, or there is an issue with the interpretation of the register values.

I noticed in this issue that you previously mentioned fixes for 32-bit values, so I wondered whether these heat-produced counters could be affected by the same issue.

I'd particularly like to establish:

* Which Modbus registers should be used for compressor heat produced, additional heat produced and hot-water heat produced on an F1245?
* Whether these are 32-bit values and what scaling/word order they use.
* Whether the registers need to be explicitly selected/configured using NIBE Modbus Manager.
* Whether there is a way to inspect the raw register values through NibeGW's UDP interface.

I'd prefer to diagnose this using the existing NibeGW/RS485 setup rather than adding another physical Modbus connection.

Any guidance would be greatly appreciated. Thanks for the excellent work on NibeGW.

elupus, 2026-09-08:
The heat produced data requires your pump to have flow meters installed which pretty much only pumps in Finland have. So thesd registers are almost always wrong.

Hami5h, 2026-09-08:
Ah! That explains it. I've chasing my tail for a few days trying to figure out what's going on.
Anyway, thanks very much for the quick response, that's really helpful information, and explains why those heat-produced values have been behaving strangely. I appreciate you taking the time to clarify it.

I'm using an F1245-8 in the UK, so no flowmeters unfortunately.

So, I already have NibeGW successfully exposing quite a lot of useful data to Home Assistant. For example, I'm currently collecting:

* Degree Minutes
* Heat Curve / Heat Offset
* Minimum and maximum supply temperatures
* Additional Heat Runtime
* Brine In / Out temperatures
* Calculated Supply Temperature
* Compressor HW Runtime
* Compressor Runtime
* Compressor Start Count
* Compressor Status
* DHW Load and Top temperatures
* Heating Priority
* HW Additional Heat Runtime
* Outdoor Temperature
* Return Temperature
* Supply Temperature

I also have electrical consumption of the heat pump being measured independently with a Shelly CT, so I'm interested in seeing whether I can use the NIBE data together with the electrical consumption to calculate a useful COP/performance figure.

Are there any other registers you would particularly recommend for an F1245 without flow meters that would help with this?

In particular, I'm interested in:

* compressor operation/frequency or output
* heating and DHW operating information
* any reliable energy consumption registers
* anything else that would be useful for calculating or estimating COP

Also, are there any registers that you know are particularly useful for monitoring an F1245 which aren't currently exposed by the Home Assistant integration?

Thanks again for your help. I can stop messing around with those heat produced registers now.

elupus, 2026-09-08:
So you can estimate the consumption pretty well. Conpressor frequency plus its max rating, same with tye flow pumps. But production is pretty much not possible without know the flow rate. I dont think the pumps are constant displacement pumps.

Hami5h, 2026-09-08:
Yes, you're right with the data I already have I should be able to figure out an approximate performance.
Thanks very much for your help, much appreciated.

malcej1o, 2026-10-12:
I have an SMO 20 and an F2040; is there any way to use them on an RS485 bus? They don't support Modbus... I'd appreciate some advice—I've been trying to figure this out for three months now with no success.

elupus, 2026-10-13:
The SMO20 is strange. See https://github.com/yozik04/nibe/issues/41. They list parameters for it, but there is no way to enable the modbus40 accessory in the firmware. The rs485 bus is there and in theory one could get spy on other accessory data, but we depend on the modbus40 accessory setting for most data. 

So i think you are somewhat out of luck.