elupus/esphome-nibe #61: Disabled configuaration entities in HomeAssistant

elbartl, 2024-01-11:
Hello,

I have managed to successfully setup communication between my HA and Nibe using NibeGW with esphome. 
There are two entities in HA which are visible: "Hot Water" and "Climate System S1". The rest of the entities are all unavailable. My config is based on the yaml from example folder. The goal here is to coordinate heat pump with solar production. I would appreciate help with enabling those entities in HA. Maybe that will be beneficial for others too.

Setup: SMO 40 with Firmware: 9586 and TTGO T-CAN485 ESP32

Regards 
Bart

[image]

All entities are unavailable like that:
[image]

elupus, 2024-01-11:
Go to configuration of that entity and enable it.

elbartl, 2024-01-12:
Wow that works, thanks. Did not think it is that easy. Appreciate your work and quick help!

closed 2024-01-12 by elbartl