elupus/esphome-nibe #63: Request for help with wiring

Fredy72, 2024-02-02:
Here it is written about connecting the power supply directly from the CTS602, where, among other things, +12V and GND are output.
> https://github.com/nic6911/ESP32_Modbus_Module/raw/main/electronics/connection.png

Here is a picture of the LILYGO® T-CAN485 wiring in the Nibe heat pump, where PIN9 = GND and PIN12 = +12V (PINs 10 and 11 are Modbus).
> https://github.com/elupus/esphome-nibe/blob/master/examples/lilygo-t-can485.png

I own a LILYGO® T-CAN485 and would like to wiring it up to the CTS602 and powered directly from the CTS602 to the PINs 1,2,3,6.
Pin1 = 12 VDC output
Pin2 = COM1 - RS 485 Data+ (A) - Modbus
Pin3 = COM1 - RS 485 Data- (B) - Modbus
Pin6 = GND

The electrical diagram of the Lily device is as follows.
> https://github.com/Xinyuan-LilyGO/T-CAN485/blob/main/doc/schematic.pdf

I want to ask if it is possible to use the connection in Nilan as pictured with Nibe, i.e. the GND on the RS485 connector (middle PIN) would not be connected (it is written in the first link: "_but keep minimum 1x GND on Nilan in all cases!_")?
Thank you for your answer.

elupus, 2024-02-02:
What is Nilan?

Fredy72, 2024-02-02:
Hi, it is Ventilation with passive & active heat recovery.
https://www.en.nilan.dk/Files/Files/Engelsk/Downloads/6.%20Wiring%20diagrams/CTS602%20HMI%20Combi%20302%20Top%20701122%20V1-08%20231206(U).pdf

Fredy72, 2024-02-02:
Thank you for your response.
So are any of these options correct?

In the picture at Nibe it is like this:
[image]

As an alternative, I thought of this:
[image]

elupus, 2024-02-03:
I think that should be fine.

elupus, 2024-02-03:
I think you can connect both grounds. They are internally linked with 1 mega ohm resistors. But your suggestion is probably the best first bet. 

That said. I have no idea whatsoever if the nilan speaks nibe protocol (rs485 is not standardized on that level)

Fredy72, 2024-02-03:
Thank you, it's not about the protocol itself, it's fine and communication via Modbus with Nilan has been tested by others.

elupus, 2024-02-03:
Nibe does not use modbus though. Its their own protocol. We have no database for the CTS602, so the esphome nibe component is unlikely to be usefull.

elupus, 2024-02-03:
Right, but this is a completely different component. Im going to close this now. I thought you were intending to use this component. Try to open such issues in the right component. I hope it works out though.

closed 2024-02-03 by elupus