elupus/esphome-nibe #64: Unable to connect to the NibeGw

Palerm0, 2024-02-24:
Hi,
I had an NibePi installed. But I want to switch to this method. I received an LilyGo T-CAN485 last week.
I followed the steps like "Step-by-step configuration #56". This is clearly explained.
 
Now the Nibe Heatpump is connected to the gateway because the Heatpump does not fall into the error mode (Red coloured bar at the front)

But the device is not connected in ESPHome. uploading all went fine. 

This is my configuration:
```
esphome:
  name: nibegw
  comment: Nibegw
  platform: ESP32
  board: esp32dev

# General ESPHome setup
api:
  reboot_timeout: 0s

ota:
  password: "OTP"

logger:

wifi:
  ssid: "Wifi SSID name"
  password: "Wfi Pasphrase"
  fast_connect: true # Fast connect to connect to my hidden network
  # It is recommended to disable powersave mode on wifi, to make sure the device does not miss UDP requests sent.
  power_save_mode: none
  # The device needs a static IP. Either do that here with the manual_ip node, or do it via another way (router)

# Load nibe component
external_components:
  - source: github://elupus/esphome-nibe

# Set pins required for LilyGo T-CAN485 board
output:
  - platform: gpio
    id: ENABLE_PIN # Enable the chip
    pin:
      number: GPIO19
      inverted: true
  - platform: gpio
    id: SE_PIN # Enable autodirection
    pin:
      number: GPIO17
      inverted: true
  - platform: gpio
    id: ENABLE_5V_PIN # Enable 5V pin for RS485 chip
    pin:
      number: GPIO16
      inverted: true

# Configure uart that will be used
uart:
  rx_pin: GPIO21
  tx_pin: GPIO22
  baud_rate: 9600

# Configure NibeGW
nibegw:
  udp:
    # The target address(s) to send data to. May be a multicast address.
    # When using Home Assistant: this is your Home Assistant IP.
    target:
      - ip: 192.168.100.xxx #HA
        port: 9999

    # List of source address to accept data from, may be empty for no filter
    source:
      - 192.168.25.xxx #Gateway (Router)
      - 192.168.100.xxx #Gateway (Router)
      - 192.168.100.230 #HA

    # Optional port this device will listen to to receive read requests. Defaults to 9999
    #read_port: 9999

    # Optional port this device will listen to to receive write request. Defaults to 10000
    #write_port: 10001

  acknowledge:
    - MODBUS40

  # Constant replies to certain requests cabe made
  constants:
    - address: MODBUS40
      token: ACCESSORY
      data: [
            0x0A, # MODBUS version low
            0x00, # MODBUS version high
            0x01, # MODBUS address?
      ]

# Some helper functions to restart ESPHome from HA
button:
- platform: restart
  name: Nibegw Restart
- platform: safe_mode
  name: Nibegw Safe Mode Boot

```
When I open the logging of the (offline) Device in the list It shows the following:

```
INFO ESPHome 2024.2.0
INFO Reading configuration /config/esphome/lilygo-t-can485-nibe.yaml...
INFO Starting log output from nibegw.local using esphome API
WARNING Can't connect to ESPHome API for nibegw.local: Error resolving IP address: [Errno -5] No address associated with hostname (APIConnectionError)
INFO Trying to connect to nibegw.local in the background
```
Probably because the NibeGW is on a different subnet network (Wifi network). HA is on an other network because it is an VM server.
Is it possible not to connect to the nibegw.local. but to the IP address of the Nibegw? This may be solves my problem.

Thanks in advanced
Greetings
Palermo

Palerm0, 2024-03-27:
If installed an new instance of homeassistant in the same (WiFi) subnet.  
There seems it to be working. 

So it is not able to route the NibeGW to an another network (Subnet).

elupus, 2024-03-27:
That seems to be a esphome related question. I think you need mdns working across subnets. But i also think you can set a fixed ip as here https://esphome.io/components/wifi.html#manual-ips and that will be used to connect instead of mdns.