elupus/esphome-nibe #67: NIIBE VVM 320 HW Production

zejdajan, 2024-05-11:
Hi,
I have a issue with this entity: HW Production, it shows as a switch in HA.
I am using ESP32 with RS485 converter with ESP Home. VVM320+F2040, all running few years.
Everything is working- changing water options, heating temps etc...
However I suppose that this entity should turn off Hot water production, until turning it back on.
If I flip that switch, after few seconds it gets ON itself. 
Is this normal behaviour and I should change something in heat pump setting?
Is there any bug causing that?

Thank you

elupus, 2025-03-26:
It's marked as writeable in the docs by Nibe so we allow it to be changed. Unsure what effect it has to be honest. It's not an issue in the gw device. Maybe it should be adjusted to not be writable in the database at https://github.com/yozik04/nibe. But don't like doing that unless we know why it's being rejected.

Closing this here, since it's not a bug in the gateway device.

closed 2025-03-26 by elupus

zejdajan, 2025-03-28:
Ok, thank you. I created a bug report on the original repository, hopefully it will be fixed.