elupus/esphome-nibe #68: Issue with connecting to nibe f370

Mellbin, 2024-08-19:
Not sure what I am doing wrong/what is breaking here.

Have setup lilygo espboard following #56 
HP is connected and enabled.
I can see when logging wireless that the nibe integration is trying to connect but it just gives me an error. (async UDP)

HP also goes into alarm if that helps.

```
INFO ESPHome 2024.7.3
INFO Reading configuration /config/esphome/nibef370.yaml...
INFO Starting log output from 192.168.1.209 using esphome API
INFO Successfully connected to nibef370 @ 192.168.1.209 in 0.006s
INFO Successful handshake with nibef370 @ 192.168.1.209 in 0.098s
[23:42:45][I][app:100]: ESPHome version 2024.7.3 compiled on Aug 19 2024, 23:37:50
[23:42:45][C][wifi:599]: WiFi:
[23:42:45][C][wifi:427]:   Local MAC: xxxxx
[23:42:45][C][wifi:432]:   SSID: xxxxxx
[23:42:45][C][wifi:435]:   IP Address: 192.168.1.209
[23:42:45][C][wifi:439]:   BSSID: xxxxxxxx
[23:42:45][C][wifi:440]:   Hostname: 'nibef370'
[23:42:45][C][wifi:442]:   Signal strength: -55 dB ▂▄▆█
[23:42:45][C][wifi:446]:   Channel: 6
[23:42:45][C][wifi:447]:   Subnet: 255.255.255.0
[23:42:45][C][wifi:448]:   Gateway: 192.168.1.1
[23:42:45][C][wifi:449]:   DNS1: 192.168.1.1
[23:42:45][C][wifi:450]:   DNS2: 0.0.0.0
[23:42:45][C][logger:185]: Logger:
[23:42:45][C][logger:186]:   Level: DEBUG
[23:42:45][C][logger:188]:   Log Baud Rate: 115200
[23:42:45][C][logger:189]:   Hardware UART: UART0
[23:42:45][C][uart.arduino_esp32:151]: UART Bus 1:
[23:42:45][C][uart.arduino_esp32:152]:   TX Pin: GPIO22
[23:42:45][C][uart.arduino_esp32:153]:   RX Pin: GPIO21
[23:42:45][C][uart.arduino_esp32:155]:   RX Buffer Size: 256
[23:42:45][C][uart.arduino_esp32:157]:   Baud Rate: 9600 baud
[23:42:45][C][uart.arduino_esp32:158]:   Data Bits: 8
[23:42:45][C][uart.arduino_esp32:159]:   Parity: NONE
[23:42:45][C][uart.arduino_esp32:160]:   Stop bits: 1
[23:42:45][C][gpio.output:010]: GPIO Binary Output:
[23:42:45][C][gpio.output:011]:   Pin: GPIO19
[23:42:45][C][gpio.output:010]: GPIO Binary Output:
[23:42:45][C][gpio.output:011]:   Pin: GPIO17
[23:42:45][C][gpio.output:010]: GPIO Binary Output:
[23:42:45][C][gpio.output:011]:   Pin: GPIO16
[23:42:45][C][restart.button:017]: Restart Button 'Nibegw Restart'
[23:42:45][C][restart.button:017]:   Icon: 'mdi:restart'
[23:42:45][C][safe_mode.button:024]: Safe Mode Button 'Nibegw Safe Mode Boot'
[23:42:45][C][safe_mode.button:024]:   Icon: 'mdi:restart-alert'
[23:42:45][C][captive_portal:088]: Captive Portal:
[23:42:45][C][nibegw:108]: NibeGw
[23:42:45][C][nibegw:114]:  Target: 192.168.1.166:9999
[23:42:45][C][nibegw:120]:  Read Port: 9999
[23:42:45][C][nibegw:121]:  Write Port: 10000
[23:42:45][C][mdns:116]: mDNS:
[23:42:45][C][mdns:117]:   Hostname: nibef370
[23:42:45][C][esphome.ota:073]: Over-The-Air updates:
[23:42:45][C][esphome.ota:074]:   Address: nibef370.local:3232
[23:42:45][C][esphome.ota:075]:   Version: 2
[23:42:46][C][esphome.ota:078]:   Password configured
[23:42:46][C][safe_mode:018]: Safe Mode:
[23:42:46][C][safe_mode:020]:   Boot considered successful after 60 seconds
[23:42:46][C][safe_mode:021]:   Invoke after 10 boot attempts
[23:42:46][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[23:42:46][C][api:139]: API Server:
[23:42:46][C][api:140]:   Address: nibef370.local:6053
[23:42:46][C][api:142]:   Using noise encryption: YES
[23:42:46][D][nibegw:025]: UDP Packet with 6 bytes to send
[23:42:46][D][nibegw:025]: UDP Packet with 6 bytes to send
[23:42:47][D][nibegw:025]: UDP Packet with 6 bytes to send
[... 123 lines cut ...]
[23:43:24][D][nibegw:025]: UDP Packet with 6 bytes to send
[23:43:24][D][nibegw:094]: Constant to address: 0x20 token: 0xee bytes: 7
[23:43:24][D][nibegw:025]: UDP Packet with 6 bytes to send
[23:43:24][D][nibegw:094]: Constant to address: 0x20 token: 0xee bytes: 7
[23:43:24][D][nibegw:025]: UDP Packet with 6 bytes to send
[23:43:24][D][nibegw:094]: Constant to address: 0x20 token: 0xee bytes: 7
[23:43:24][D][nibegw:025]: UDP Packet with 6 bytes to send
[23:43:24][D][nibegw:094]: Constant to address: 0x20 token: 0xee bytes: 7
[23:43:25][D][nibegw:048][async_udp]: UDP Packet token data of 6 bytes received
[23:43:26][D][nibegw:025]: UDP Packet with 6 bytes to send
[23:43:26][D][nibegw:025]: UDP Packet with 6 bytes to send
[23:43:27][D][nibegw:025]: UDP Packet with 6 bytes to send
[23:43:27][D][nibegw:025]: UDP Packet with 6 bytes to send
[23:43:27][D][nibegw:085]: Response to address: 0x20 token: 0x69 bytes: 6
[23:43:27][D][nibegw:025]: UDP Packet with 6 bytes to send
[23:43:27][D][nibegw:025]: UDP Packet with 6 bytes to send
[23:43:28][D][nibegw:025]: UDP Packet with 6 bytes to send
[23:43:28][D][nibegw:025]: UDP Packet with 6 bytes to send
[23:43:29][D][nibegw:025]: UDP Packet with 6 bytes to send
[23:43:29][D][nibegw:025]: UDP Packet with 6 bytes to send
[23:43:29][D][nibegw:025]: UDP Packet with 6 bytes to send
[23:43:29][D][nibegw:025]: UDP Packet with 6 bytes to send
[23:43:30][D][nibegw:048][async_udp]: UDP Packet token data of 6 bytes received

```

elupus, 2024-08-20:
Not sure what might go wrong.. Can you keep that log window open when you try to add it in HA? 

What is your config in HA?

Mellbin, 2024-08-20:
The logs included are what I see when i try to connect the nibegw integration.
It sends 3 of these requests but i don't see any answer to them. then the integration shows the warning that it's the wrong adress/port
[23:43:20][D][nibegw:048][async_udp]: UDP Packet token data of 6 bytes received

Which config are you referring to?

elupus, 2024-08-20:
The esphome config file, and a screenshot of your config attempt in HA

Mellbin, 2024-08-20:
Config file:
```
esphome:
  name: nibef370
  friendly_name: NIBEF370

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "nwnpht/2bA/rkNApMAy5ZVgbs8Ied9oBJxQGXgnqbOc="

ota:
  - platform: esphome
    password: "aaae23bae4650a6299bd999b9f4a2081"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true # Fast connect to connect to my hidden network
  # It is recommended to disable powersave mode on wifi, to make sure the device does not miss UDP requests sent.
  power_save_mode: none
  # The device needs a static IP. Either do that here with the manual_ip node, or do it via another way (router)

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Nibef370 Fallback Hotspot"
    password: "BFLlIBAvUMHf"

captive_portal:

  # Load nibe component
external_components:
  - source: github://elupus/esphome-nibe

# Set pins required for LilyGo T-CAN485 board
output:
  - platform: gpio
    id: ENABLE_PIN # Enable the chip
    pin:
      number: GPIO19
      inverted: true
  - platform: gpio
    id: SE_PIN # Enable autodirection
    pin:
      number: GPIO17
      inverted: true
  - platform: gpio
    id: ENABLE_5V_PIN # Enable 5V pin for RS485 chip
    pin:
      number: GPIO16
      inverted: true

# Configure uart that will be used
uart:
  rx_pin: GPIO21
  tx_pin: GPIO22
  baud_rate: 9600

# Configure NibeGW
nibegw:
  udp:
    # The target address(s) to send data to. May be a multicast address.
    # When using Home Assistant: this is your Home Assistant IP.
    target:
      - ip: 192.168.1.166
        port: 9999 #The Nibe Home Assistant integration listens to 9999 by default 

    # List of source address to accept data from, may be empty for no filter
    source:
      #- 192.168.1.166
    #read_port: 10081   
   # write_port: 10082

  acknowledge:
    - MODBUS40

  # Constant replies to certain requests cabe made
  constants:
    - address: MODBUS40
      token: ACCESSORY
      data: [
            0x0A, # MODBUS version low
            0x00, # MODBUS version high
            0x01, # MODBUS address?
      ]

# Some helper functions to restart ESPHome from HA
button:
- platform: restart
  name: Nibegw Restart
- platform: safe_mode
  name: Nibegw Safe Mode Boot
```

Screenshot with modbus disabled on heatpump.
[image]

Screenshot with modbus enabled on heatpump.
[image]

elupus, 2024-08-20:
Remote address in that screenshot seem empty? Should be the IP of the esphome device (maybe it got cleared on error?)

How is your home assistant installed? HAOS or some custom set up? (Need to accept reception of udp packets)

Mellbin, 2024-08-20:
Yes, it clears on error.

I'm using HAOS on raspberry pi 5.

Mellbin, 2024-08-20:
Hmmm, decreased the log level to warn but did not seem to help.

Currently got the board revision v1.1 (2024-04-25).
Don't know if anyone has confirmed this one to work? but should be similar enough?

elupus, 2024-08-20:
The most common issue is with write commands not triggering write commands on the bus. I dont know if that chip has automatic switch between read and write (which the config assume)

Mellbin, 2024-08-20:
What i gathered from checking the v1.0 board and mine is that the [rs485 chip](https://www.analog.com/en/products/max13487e.html#part-details) is the same 

What do you think is the next step?

elupus, 2024-08-25:
You could try the new release. I did some performance improvements for the auto rx/tx type chips. Not sure it is your problem.

Mellbin, 2024-09-02:
Hi again,

So after a while I felt like working on this again and not sure what happened since the last time i tried but now when I try to get the HP to talk with the board i get this error. (Sorry for HDR screenshot)
[image]

elupus, 2024-09-02:
Yea that is some new logging added. It very much looks like the pump is not seeing the things sent by the GW device at all. Sort of looks like the auto rx/tx function is not working on the device. Either config for that board is wrong. Board could be broken, or it could be wired up incorrectly.

Hmm. I seem to recall some issue with grounding on some boards.

Mellbin, 2024-09-02:
Since I've tried 2 different boards (ordered 2 at the same time) it should not be the board unless there is some manufacturing issue?

I'm also guessing you can't see any issues on the config for the board that I sent?

So grounding may be the issue, any link to a discussion regarding this?

elupus, 2024-09-02:
It was this issue: https://github.com/elupus/esphome-nibe/issues/46 but looks like that got that resolved.

Mellbin, 2024-09-10:
I prob had warning log level in the last image, sorry for the confusion...

But when turning on the verbose logging it very much looks like the controller is talking to the HP.
Maybe you can gather something from these logs, otherwise I will give up with the lilygo board...

[logs_nibev2_logs2.txt](https://github.com/user-attachments/files/16948151/logs_nibev2_logs2.txt)

elupus, 2024-09-10:
All the: "Unexpected Ack/Nack: 5C" means the heatpump never receive what the nibegw sends.

Something is wrong with the transmission path to the heatpump. Maybe cabling, connections (though unlikely since we do receive data). Hardware, or pin setup. I cant really say without hooking up a logic analyser or oscilloscope to the cabling.

Mellbin, 2024-09-11:
Yeah so I've now given up on this board.
Have ordered the other recommended board [T-RS S3](https://www.lilygo.cc/products/t-rs-s3?_pos=5&_sid=d6e515864&_ss=r) to test instead.

Do you want me to close this issue now? Or should I try the other board first?
Ff the t-rs-s3 board works I will make a PR with example :)

elupus, 2024-09-11:
Leave it open till you tested that board. Note that that board does not support auto rx/tx enabling. So you need to hook up dir_pin config on that one.

Mellbin, 2024-09-19:
It's not getting better...

Can you see something off here?

[image]

```
esphome:
  name: nibev3
  friendly_name: nibev3

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino

# Enable logging
logger:
  level: VERY_VERBOSE

# Enable Home Assistant API
api:
  encryption:
    key: "18iUkMNxkAbVF2PJgBWNVjRcN/35yFSvNQYOztFC/pc="

ota:
  - platform: esphome
    password: "1bcb0d28b86c99451fa369acb32728ae"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true # Fast connect to connect to my hidden network
  # It is recommended to disable powersave mode on wifi, to make sure the device does not miss UDP requests sent.
  power_save_mode: none
  # The device needs a static IP. Either do that here with the manual_ip node, or do it via another way (router)

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Nibev3 Fallback Hotspot"
    password: "HJimn3uiSSRW"

captive_portal:

   # Load nibe component
external_components:
  - source: github://elupus/esphome-nibe@v3.0.1
    components: [ nibegw ]
    refresh: 0s

# Configure uart that will be used
uart:
  id: qwerty
  rx_pin: GPIO2
  tx_pin: GPIO3
  baud_rate: 9600

output:
  - platform: gpio
    id: LED_PIN
    pin:
      number: GPIO1
      inverted: True
    

# Configure NibeGW
nibegw:

  uart_id: qwerty

  dir_pin:
    number: GPIO4
    inverted: false

  udp:
    # The target address(s) to send data to. May be a multicast address.
    # When using Home Assistant: this is your Home Assistant IP.
    target:
      - ip: 192.168.1.166
        port: 9999 #The Nibe Home Assistant integration listens to 9999 by default 

    # List of source address to accept data from, may be empty for no filter
    source:
      - 192.168.1.166
    #read_port: 10081   
   # write_port: 10082

  acknowledge:
    - MODBUS40

  # Constant replies to certain requests cabe made
  constants:
    - address: MODBUS40
      token: ACCESSORY
      data: [
            0x0A, # MODBUS version low
            0x00, # MODBUS version high
            0x01, # MODBUS address?
      ]

# Some helper functions to restart ESPHome from HA
button:
- platform: restart
  name: Nibegw Restart
- platform: safe_mode
  name: Nibegw Safe Mode Boot
    
```

[logs_nibev3_run.txt](https://github.com/user-attachments/files/17063855/logs_nibev3_run.txt)

elupus, 2024-09-19:
Very verbose usually don't work. Also log seem to be cut to short. But otherwise nothing obvious. 

I would recheck cabling and wich pins you use.

Mellbin, 2024-10-15:
Since I've tried multiple boards, all kinds of configs, multiple cables and both powered the cards via USB-C and the HP 12v supply and nothing works I will put the blame on the actual heat pump not working as intended.

Would be nice to verify with an actual MudBus40 unit but don't have one available.

Thanks for the help @elupus!

closed 2024-10-15 by Mellbin

elupus, 2024-10-15:
Sad to hear. It mostly sound like a cabling problem.

Mellbin, 2024-10-15:
Seems unlikely since I've tried a TP CAT cable, USB cable and also normal "FK" cables as communication cables. :(