elupus/esphome-nibe #76: Esphome won't compile with virtual RMU climate component added

svollebregt, 2024-08-31:
I wanted to try out the virtual RMU option as this has been something I was looking for, but I get the following error when building:

```
INFO ESPHome 2024.8.1
INFO Reading configuration nibe-noeme.yaml...
Failed config

climate.nibegw: [source nibe-noeme.yaml:169]

  'system' is a required option for [climate.nibegw].
  platform: nibegw
  name: s3

  [address] is an invalid option for [climate.nibegw]. Please check the indentation.
  address: RMU40_S3
  sensor: current_temperature_s3
```

Below is my configuration, which builds fine with the climate: part  commented out:
```
esphome:
  name: nibe
  friendly_name: nibe

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  reboot_timeout: 0s
  encryption:
    key: ""

ota:
  - platform: esphome
    password: ""

wifi:
  ssid: 
  password: 
  power_save_mode: none

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Nibe Fallback Hotspot"
    password: ""

captive_portal:
    
# Load nibe component
external_components:
  - source: 
      type: git
      url: https://github.com/elupus/esphome-nibe.git
    components: [ nibegw ]
    
# Set pins required for LilyGo T-CAN485 board
output:
  - platform: gpio
    id: ENABLE_PIN # Enable the chip
    pin:
      number: GPIO19
      inverted: true
  - platform: gpio
    id: SE_PIN # Enable autodirection
    pin:
      number: GPIO17
      inverted: true
  - platform: gpio
    id: ENABLE_5V_PIN # Enable 5V pin for RS485 chip
    pin:
      number: GPIO16
      inverted: true

# Configure uart that will be used
uart:
  rx_pin: GPIO21
  tx_pin: GPIO22
  baud_rate: 9600

# Configure NibeGW
nibegw:
  udp:
    # The target address(s) to send data to. May be a multicast address.
    target:
      - ip: 192.168.1.xyz
        port: 9999

    # List of source address to accept data from, may be empty for no filter
    source:
      - 192.168.1.xyz
      
    read_port: 9999
    write_port: 10000

  acknowledge:
    - MODBUS40

  # Constant replies to certain requests cabe made
  constants:
    - address: MODBUS40
      token: ACCESSORY
      data: [
            0x0A, # MODBUS version low
            0x00, # MODBUS version high
            0x01, # MODBUS address?
      ]

# Some helper functions to restart ESPHome from HA
button:
- platform: restart
  name: Nibegw Restart
- platform: safe_mode
  name: Nibegw Safe Mode Boot

# I2C
i2c:
  sda: 32
  scl: 33
  scan: true
  id: bus_a
 
# HT21d 
sensor:
  - platform: htu21d
    temperature:
      name: "TR Temperature"
    humidity:
      name: "TR Humidity"
    update_interval: 60s
  - platform: homeassistant
    id: current_temperature_s3
    entity_id: sensor.tfa_ts34c_cresta_5f_0e_temperature

#  MCP23017
mcp23017:
  - id: 'mcp23017_hub'
    address: 0x20

# Relais for bathroom heating loop    
switch:
  - platform: gpio
    pin: 
      mcp23xxx: mcp23017_hub
      # Pin B0
      number: 8
      mode:
        output: true
      inverted: true
    name: "Bathroom heating"
    
  - platform: gpio
    pin: 
      mcp23xxx: mcp23017_hub
      # Pin B1
      number: 9
      mode:
        output: true
      inverted: true
    name: "Nibe water luxe"
    
  - platform: gpio
    pin: 
      mcp23xxx: mcp23017_hub
      # Pin B2
      number: 10
      mode:
        output: true
      inverted: true
    name: "Nibe block heating"
    
  - platform: gpio
    pin: 
      mcp23xxx: mcp23017_hub
      # Pin B3
      number: 11
      mode:
        output: true
      inverted: true
    name: "Nibe block cooling"
    
# Add a virtual RMU on S3
climate:
  - platform: nibegw
    name: s3
    address: RMU40_S3
    sensor: current_temperature_s3
```
It may be something trivial, but I cannot find the solution. I am running the latest ESPHome (2024.8.1) and performed a cleanup of the build configuration. As mentioned, without the last climate part, it compiles fine.

elupus, 2024-08-31:
Sorry, i forgot to update docs. Replace address: RMU_S3 with system: 3

svollebregt, 2024-09-01:
Thanks. It now compiles but in the log it states:
`[I][nibegw:155]: Publishing to rmu: 0x1b temp: nan -> 00.80`

Earlier on when initializing:
```
[10:08:52][C][api:139]: API Server:
[10:08:52][C][api:140]:   Address: nibe.local:6053
[10:08:52][C][api:142]:   Using noise encryption: YES
[10:08:52][C][homeassistant.sensor:030]: Homeassistant Sensor 'current_temperature_s3'
[10:08:52][C][homeassistant.sensor:030]:   Device Class: 'temperature'
[10:08:52][C][homeassistant.sensor:030]:   State Class: 'measurement'
[10:08:52][C][homeassistant.sensor:030]:   Unit of Measurement: '°C'
[10:08:52][C][homeassistant.sensor:030]:   Accuracy Decimals: 1
[10:08:52][C][homeassistant.sensor:031]:   Entity ID: 'sensor.tfa_ts34c_cresta_5f_0e_temperature'
```

The sensor is defined like this, I first tried the minimal example without the state and device class as in your example but got the same result:
```
sensor:
  - platform: htu21d
    temperature:
      name: "TR Temperature"
    humidity:
      name: "TR Humidity"
    update_interval: 60s
  - platform: homeassistant
    id: current_temperature_s3
    unit_of_measurement: "°C"
    device_class: "temperature"
    state_class: "measurement"
    entity_id: sensor.tfa_ts34c_cresta_5f_0e_temperature
```
The sensor itself works in HASS and has a valid number, so I am puzzled as to why the ESP nodes make it NaN.

elupus, 2024-09-01:
Right. Then switch the climate entity to heat cool

svollebregt, 2024-09-01:
Climate system S3 does not show up in HASS. When I force enable it, it throws an error that it does not exist. Do all pumps support multiple climate systems, mine is a F1255 if that is relevant

elupus, 2024-09-01:
It shows up under the esphome device, not the nibe integrationen. This is not really a good solution i agree

svollebregt, 2024-09-01:
Ok, how do I control the state to HEAT_COOL? I see I need to define an action, but if I try to do it on_boot: or on_time_sync:, it won't compile.

elupus, 2024-09-01:
You control that easiest from home assistant.

svollebregt, 2024-09-01:
Ah, found it. This indeed works!

closed 2024-09-01 by svollebregt