elupus/esphome-nibe #77: No issue, but a question about nibegw virtual RMU

OnlyRobin, 2024-09-03:
Everythings works fine but the sensor that gives the temp is 25.350000 and what its Publishing to rmu: 0x19 temp: 25.350000 -> 05.01, so with 0.5 more.
And what I get in my heat pump is 25.4

I there a way to get the same value?

I'ts not nessery but it looks nicer :-)

elupus, 2024-09-03:
Please check on the pump menu. I think there is a .5 offset in what the rmu sends.

OnlyRobin, 2024-09-03:
> Please check on the pump menu. I think there is a .5 offset in what the rmu sends.

Hi,

No i have checked that, offset is 0. 
If i change it nothing happens.

elupus, 2024-09-03:
Not that offset. I mean what room temp is the pump showing when we send a virtual room temp.

OnlyRobin, 2024-09-03:
This is the log:
[18:55:51][I][nibegw:155]: Publishing to rmu: 0x19 temp: 25.280001 -> 04.01
[18:55:52][I][log:144]: 25.280001
[18:55:54][D][climate:396]: 'verwarming woonkamer temperatuur RMU40 S1' - Sending state:
[18:55:54][D][climate:399]:   Mode: HEAT_COOL
[18:55:54][D][climate:419]:   Current Temperature: 25.30°C
[18:55:54][D][climate:425]:   Target Temperature: 19.00°C

And this is the temp the pump is showing: Current Temperature: 25.30°C

elupus, 2024-09-03:
So that looks like it matches then.

elupus, 2024-09-03:
The idea is that it should round to nearest tenth of a degree. 25.44 aught to be 25.4, while 24.45 would be 24.5. 

But we might be rounding wrong somewhere.

OnlyRobin, 2024-09-03:
Now it's rounding good from 25.04 to 25 so thats good.

closed 2024-09-07 by OnlyRobin