elupus/esphome-nibe #80: Error when resetting alarm

frma71-2, 2024-11-03:
Since I upgraded my Nibe F1255-12 to firmware 9699R7 I get an error when trying to reset an alarm.

The error is: "Heatpump denied writing alarm-reset-45171"

elupus, 2024-11-03:
Mostly a duplicate of https://github.com/home-assistant/core/issues/120023 as you already found. 

This looks to be a firmware bug on the pumps. Nothing that we can solve. Would need somebody with an actual modbus 40 device to report this to nibe

frma71-2, 2024-11-03:
Yes, but I though since it's different projects and I assume the problem is closer to eshome-nibe then home-assistant/core I reported it here as well. Sorry if I messed up that way.

Does anyone with a premium subscription know if it's possible to reset from the myUplink UI ?

jmrantan, 2024-12-06:
Yes, resetting from the myUplink UI works perfectly (as does the setting of the pump to the ”aid” mode instead of resetting the alarm).

Any idea if you could reset the alarms using the myUplink integration? I have not had a closer look at that yet.

Skinza, 2025-01-02:
I am indeed facing the same issue.
What i noticed is that when i power cycle the heatpump i can reset the alarm once via the integration/modbus.
After that the issue comes back.

LukasdeBoer, 2025-03-22:
Same problem here after upgrading SMO40 to 9707R4..

jmrantan, 2025-03-24:
I recently updated my F1255 with firmware 9714R2, the release notes of which refer to some fixes for the resetting of the alarms.

After the update, the ”reset alarm” button in the HA integration still threw me the ”heatpump denied writing alarm-reset-45171” error, but I did subsequently find something that might be of interest.

For some reason, the ”number.alarm_reset_45171” entity had been disabled (I have not done anything to that entity ever). I am not sure if it is supposed to be/remain disabled, but I did nevertheless enable it, and thereafter I have succeeded in resetting the alarm (on two separate occasions) by moving the slider of that entity all the way to 255 and then clicking the reset button. For me, these two actions are needed (in this sequence), as neither of them works separately.

Could this be worth looking into?

LukasdeBoer, 2025-03-24:
I can confirm, this works for me too. Nice find, @jmrantan! Once you click the existing reset button, it does also reset the number.alarm_reset_45171 back to 0.

Skinza, 2025-03-24:
Nice find @jmrantan , i can confirm as well that setting the alarm_reset_45171 to 255 & then push the Alarm Reset works.
And as @LukasdeBoer also pointed out: alarm_reset_45171 returns to 0 after an Alarm Reset

elupus, 2025-03-24:
Weirdness. Ok. That should be doable as standard.

LukasdeBoer, 2025-03-26:
I already had an automation in home assistant that would automatically reset the alarm when it triggered.

As a workaround for this issue, I've added the extra step in there and it works. You will probably(?) need to update some of the entity IDs, the first entity id is the alarm slider and the second one is the alarm reset button, which is the same as previously.
```
alias: Auto reset Nibe when alarm triggers
description: ""
triggers:
  - trigger: numeric_state
    entity_id:
      - sensor.alarm_45001
    above: 0
conditions: []
actions:
  - repeat:
      sequence:
        - device_id: de2dc416604cdf6d60e399a5b0c1ec23
          domain: number
          entity_id: 2fe63dbddc3953f4fd41f8e6030a5796
          type: set_value
          value: 255
        - device_id: de2dc416604cdf6d60e399a5b0c1ec23
          domain: button
          entity_id: bfc8db618f15d631f6d98d01477e4fc0
          type: press
        - delay:
            hours: 0
            minutes: 1
            seconds: 0
            milliseconds: 0
      until:
        - condition: numeric_state
          entity_id: sensor.alarm_45001
          below: 1
          above: -1
mode: single
```

This just auto-reset the alarm for me.

elupus, 2025-03-26:
Guys. Could you test to just set the value 255. Then wait for one minute (or two) and see if that also resets the alarm.

Skinza, 2025-03-26:
First try gave me the following findings:
- set number.alarm_reset_45171 to 255
- number.alarm_reset_45171 becomes unknown
- after waiting 1m alarm was still active
- tried to set number.alarm_reset_45171 again to 255 which gave me a "denied writing" message
- pushed the alarm reset: alarm cleared & number.alarm_reset_45171 returned to 1

elupus, 2025-03-26:
Could you try writing a zero instead of 255 and waiting?

Skinza, 2025-03-26:
Findings:
- set number.alarm_reset_45171 to 0
- number.alarm_reset_45171 to becomes 0 (does not go to unknown)
- this state remained for 5m (alarm still active)
- pushed the alarm reset: alarm cleared & number.alarm_reset_45171 returned to 1

elupus, 2025-03-26:
Ok. So works to write it to 0 first. That make a lot more sense i think.

closed 2025-12-14 by elupus