elupus/esphome-nibe #81: Nibe F1145 heat pump

scruysberghs, 2024-11-18:
Hi

I have a F1145 that I want to control using Home Assistant and esphome (or modbus). Had a look inside and there are 2 controller boards in there (one for home and one for swimming pool I think) but they don't look like any of the other wiring schema's I've seen with the number 11 and 12 being used. Am I not looking in the right place? Anyone here with more Nibe HP experience that can shed a light?

[image]
[image]

elupus, 2024-11-18:
Please look at the official modbus 40 documentation from nibe.

scruysberghs, 2024-11-18:
ok thx, I'll have another look. For future reference: 
[image]

scruysberghs, 2024-11-28:
So I have connected terminals 10 and 11 to my esp directly (don't have the modbus 40 device)
Number 10 is connected to A  and 11 to B. These correspond to pin 1 and 3 of the esp according to the specs of the device I am using.
[image]
[image]
[image]

I have this UART setup in my esphome config  
- id: my_uart
    rx_pin: 1
    tx_pin: 3
    baud_rate: 9600

The esp has ip adres 192.168.0.125, home assistant is 192.168.0.243
I see this in my esphome log :
```
[14:48:42][C][nibegw:120]: NibeGw
[14:48:42][C][nibegw:122]:  Target: 192.168.0.243:9999
[14:48:42][C][nibegw:125]:  Source: 192.168.0.243
[14:48:42][C][nibegw:127]:  Read Port: 9999
[14:48:42][C][nibegw:128]:  Write Port: 10000
```

And yet in home assistant I get:
[image]

Any tips on what to try to get this working? I have tried swapping RX and TX pins already. 
Could it be a signal quality issue? Cable is only 3 meters long.

scruysberghs, 2024-11-29:
forgot to mention the heat pump goes into alarm mode if I enable the MODBUS40 accessory in the 5.2 menu so I guess something is wrong with the connection?

scruysberghs, 2024-11-29:
my board definition in esphome is :
``` 
esp32:
  variant: esp32s3
  board: esp32-s3-devkitc-1
``` 

pin definitions:
``` 
#define _HOSTNAME             "Ultra-Dongle"
#define _DEFAULT_HOSTNAME _HOSTNAME
#define OTAURL              "http://ota.smart-stuff.nl/ultra/v5/"
#define PROFILE "[ULTRA]"

//poorten
#define IO_BUTTON             0 //multifunctional flash/button

#define RGBLED_PIN           42
#define BRIGHTNESS           50 // Set BRIGHTNESS to about 1/5 (max = 255)
#define DTR_IO               17
#define RXP1                 18
#define TXP1                 -1 //no txp1
#define TXO1                 21
#define O1_DTR_IO            15
#define P1_LED               16
#define SSL_RAND              4
#define LED              	 41 //fake 
#define PRT_LED           	 16

#define IO_S0_1             45 
#define IO_H2O_2            46
#define IO_WATER     		46 //double
#define GPIO5               5
#define GPIO6               6
#define GPIO7               7
#define GPIO8               8
#define GPIO9               9

//ETH API
#define INT_GPIO            14
#define MISO_GPIO           13
#define MOSI_GPIO           11
#define SCK_GPIO            12
#define CS_GPIO             10

//MODBUS
#define RXPIN       		1
#define TXPIN       		3
#define RTSPIN      		2
#define BAUDRATE  		 9600
``` 

What is the tx/rx selection pin supposed to do?

I am correct that on the heat pump I have to enable the MODBUS40 accessory and not the MODBUS setting?

elupus, 2024-11-29:
Rs485 chips need to know if they are transmitting or receiving, this is controlled by a pin on the chips. So it depends on what rs485 treanceiver your board is using and how it is connnected to the esp32

elupus, 2024-11-29:
Might be the RTSPIN in your case, but without looking at schematics or docs for the board its unknown. Also if its active high or low matters (but that you can test)

elupus, 2024-11-29:
This is the `dir_pin` config option in nibegw

scruysberghs, 2024-11-29:
Ok thx, I'll try with 
```
  dir_pin:
    number: 2
    inverted: false
```
and inverted true and false.

> I am correct that on the heat pump I have to enable the MODBUS40 accessory and not the MODBUS setting?

scruysberghs, 2024-11-29:
on the same esp hardware I have successfully connected to a Huawei solar modbys RTU using:
```
uart:
  - id: mod_bus
    baud_rate: 9600
    rx_pin: 1
    tx_pin: 3
    rx_buffer_size: 1024
    data_bits: 8
    stop_bits: 1
    parity: NONE

modbus:
  - uart_id: mod_bus
    id: modbus1
    flow_control_pin: 2
    send_wait_time: 200ms

modbus_controller:
  - id: huawei_inverter
    address: 0x1
    modbus_id: modbus1
    setup_priority: -10
    command_throttle: 750ms
    update_interval: 15s
```

So the flow control pin is indeed 2 and what NibeGW calls the dir_pin
Is there no need to set data_bits, stop_bits or parity in the UART section for NibeGW?

elupus, 2024-11-29:
No usually not. It should defsult to the right thing i think.

scruysberghs, 2024-11-30:
I'm still missing something obvious I'm afraid. The heatpump goes into an alarm state the second the modbus option is enabled. I have tried with the dir_pin inverted true and false

elupus, 2024-11-30:
Do you get anything logged in the esp device when you turn on modbus in the punp?

bmwmotor, 2024-12-01:
I have the F1145 as well and good working for two years.
This is my setup in NibeGW.yaml file

# Set pins required for LilyGo T-CAN485 board
output:
  - platform: gpio
    id: ENABLE_PIN # Enable the chip
    pin:
      number: GPIO19
      inverted: true
  - platform: gpio
    id: SE_PIN # Enable autodirection
    pin:
      number: GPIO17
      inverted: true
  - platform: gpio
    id: ENABLE_5V_PIN # Enable 5V pin for RS485 chip
    pin:
      number: GPIO16
      inverted: true

# Configure uart that will be used
uart:
  rx_pin: GPIO21
  tx_pin: GPIO22
  baud_rate: 9600

# Configure NibeGW
nibegw:
  udp:
    # The target address(s) to send data to. May be a multicast address.
    target:
      - ip: 192.168.2.16
        port: 10090

    # List of source address to accept data from, may be empty for no filter
    source:
      - 192.168.2.16

  acknowledge:
    - MODBUS40

  # Constant replies to certain requests cabe made
  constants:
    - address: MODBUS40
      token: ACCESSORY
      data: [
            0x0A, # MODBUS version low
            0x00, # MODBUS version high
            0x01, # MODBUS address?
      ]

scruysberghs, 2024-12-01:
Can you confirm your Home Assistant add-on settings?
That 10090 should match what is set in home assitant, no? If I use 10090 like in the file you shared I guess I should set it up like this in HA: 
choose NibeGW
[image]
[image]

@bmwmotor can you confirm my wiring (the 10 and 11 port) is ok? I use an external power supply for the esp so I do not use the 5v from the heat pump to power the esp like you seem to do.

I understand it correctly I have to get the pump connected to the esp device before I can connect the home assistant nibe integration to the esp device?

scruysberghs, 2024-12-01:
I still worries me that the the home assistant add-on has the text "Verify that The MODBUS40 accessory has been enabled in the heat pump configuration." and my heat pump only allows me to enable a MODBUS setting in it's menu.

elupus, 2024-12-01:
@scruysberghs if you get errors toggling that setting. It is the right setting.

Dont bother with home assistant settings until you can turn that setting (modbus) on without errors. It should work standalone. Turn up logging in the esphome device and look if it is receiving data and/or trying to send.

Ps. Pump supplies 12 volt, not 5.

bmwmotor, 2024-12-02:
[image]

This is how it is connected. The wires are on nr 9, 10,11 en 12.   9 and 12 are power.

bmwmotor, 2024-12-02:
You find the correct pinnumbers on the docs with the board
[image]
[image]

For the F1145 you use the RS485 connection.

bmwmotor, 2024-12-02:
In the NIBE F1145 there are 2 menu's to change:

[image]
[image]

bmwmotor, 2024-12-02:
[image]

In HA --> ESPHome go to LOGS at your heatpump. It opens a new window : click on wireless.
Then you see connection messages: UDP packet xx bit send.........  Then you know your connection works.

When not look at your router if the IP adress is correct and fixed.

When this is oke then go to the NIBE add-on in HA.  I do not remember how I filled in that ports. Just try some settings

This part in ESPHome config file is sending the messages to your HA - Nibe addon.

# Configure NibeGW
nibegw:
  udp:
    # The target address(s) to send data to. May be a multicast address.
    target:
      - ip: 192.168.2.16
        port: 10090

    # List of source address to accept data from, may be empty for no filter
    source:
      - 192.168.2.16
 

Be aware that the board stops communicating when it gets updates in HA-ESPHome. Then the F1145 gives an MODBUS error. Just cancel it on the heatpump and it will work. Good Luck

scruysberghs, 2024-12-02:
only thing in my esphome logs  is this 

[12:23:10][C][nibegw:120]: NibeGw
[12:23:10][C][nibegw:122]:  Target: 192.168.0.243:10090
[12:23:10][C][nibegw:125]:  Source: 192.168.0.243
[12:23:10][C][nibegw:127]:  Read Port: 9999
[12:23:10][C][nibegw:128]:  Write Port: 10000

when enabling the modbus option nothing is logged and the heat pump goes into alarm.
I do not have the 5.3 menu in @bmwmotor  second picture. Not even after enabling MODBUS.

bmwmotor, 2024-12-02:
Do you have Nibe menu 5.2.4?   Scroll all the way down and activate MODBUS there

scruysberghs, 2024-12-02:
my esp device ha a SP3485EN chip so I've ordered a LillyGo board to ensure that isn't the problem

scruysberghs, 2024-12-02:
> Do you have Nibe menu 5.2.4? Scroll all the way down and activate MODBUS there

Yes, that is the option that causes the alarm when activated

bmwmotor, 2024-12-02:
and when you leave it on (ignore the alarm), can you then get 5.3 ?

Here is the compleet config file. You have to change IP adress and Wifi etc .
[nibegw (1).yaml.txt](https://github.com/user-attachments/files/17977610/nibegw.1.yaml.txt)

elupus, 2024-12-02:
@scruysberghs i wonder if you have some old variant of the f1145, since your photos of the boards dont seem to match the modbus40 images.

scruysberghs, 2024-12-02:
the pictures in my first post were the wrong boards somewhere in the back of the heat pump. Quite sure the cable now is connected to the right board like in @bmwmotor 's picture. 
[image]

bmwmotor, 2024-12-02:
I found the ports in the ESPHome logfile. You need them in the HA (Nibe) add-on screens

[13:43:36][C][nibegw:122]:  Target: 192.168.2.16:10090
[13:43:36][C][nibegw:125]:  Source: 192.168.2.16
[13:43:36][C][nibegw:127]:  Read Port: 9999
[13:43:36][C][nibegw:128]:  Write Port: 10000

scruysberghs, 2024-12-02:
> Could your pump have a really old firmware version?

that could well be the case. We are trying to upgrade it now (it never had a firmware update since it was installed)

Sintyr, 2025-01-02:
Hey, how did the firmware update go? I have the same issue I cannot log any data and I have the original Modbus 40 accessory connected to F1145. Still getting errors I tried serial connection to HA with RS485 to USB now I am trying with ELFIN11RTU to TCP. Looking for a solution too

Swissgreen, 2025-01-04:
Look on page 10 of this manual, it says that the module software must be 1161 or later. Can you update through an USB key?

[https://www.nibe.fi/assets/documents/9133/031725-4.pdf](url)

scruysberghs, 2025-01-05:
> Hey, how did the firmware update go? I have the same issue I cannot log any data and I have the original Modbus 40 accessory connected to F1145. Still getting errors I tried serial connection to HA with RS485 to USB now I am trying with ELFIN11RTU to TCP. Looking for a solution too

Did not do the upgrade myself but word of warning. The owner appartly used a wrong version of the firmware, got an error message after rebooting and after using the option to revert to the previous/original firmware the controlboard was bricked and had to be replaced... Mind you, this heat pump was 15 years old and still had the original firmware.

markotanninen, 2025-01-19:
Hey, I tried reading several years back without the Modbus 40 box and gave up. I bought the Modbus 40 box and connected it and then I was able to read some of the registers, but then I ran out of time. Now I have time(priority) and am continuing. I have the Modbus40 box connected to RTU/TCP/IP converter. For some reason after I read several times, something jams up in the reading of the registers. I get a response from some of the registers, but I have been having trouble figuring out what the default Modbus register list is and if it must be programmed /if so how to do it. Anyone have documentation on this?

elupus, 2025-01-19:
@markotanninen this project does not use modbus40. however you can use the python project at https://github.com/yozik04/nibe to read from a modbus40 device. Thou there are some people that have trouble with it which can be seen here https://github.com/home-assistant/core/issues/134974. In theory that python library should work, but we have little confirmation on that.

markotanninen, 2025-01-19:
> home-assistant/core#134974

Thank you Joachim!