elupus/esphome-nibe #82: Add BT71 entry to F1155/F1255 and fix for BT74

stixif, 2024-12-04:
Hi, i become this error...

:34.084 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring coil 42439 value ecf9 - failed to decode
2024-12-04 01:01:34.084 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring coil 42447 value aaf4 - failed to decode
2024-12-04 01:01:36.322 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring coil 42437 value b17c - failed to decode
2024-12-04 01:01:36.322 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring coil 42439 value ecf9 - failed to decode

they found a solution...
https://github.com/yozik04/nibe/pull/153

can you help me to find a solution

elupus, 2024-12-04:
My guess is you added these to the LOG.SET file. Please revive them from that file. 

The pumps usually fail to send 32 bit values using log set entries.

stixif, 2024-12-04:
thanks a lot... 
this work for a long time with my FHEM installation... but now ich will change to Hass
so i searched the Problem on the other side...

so it was my mistake in the LOG.SET of the Nibe

closed 2024-12-04 by stixif