elupus/esphome-nibe #83: Writing certain value fails (checksum = 5c)

qabi, 2024-12-10:
I don't know if this is an issue in my actual heat pump controller (VVM500) or in this project?

Writing to 47032 (external adjustment) works for values 3, 2, 1, 0, -1, -3 and -4 but NOT for -2. That seems very strange.

Actual data sent:

-1: `c06b06b8b7ff0000005d` (works)
-2: `c06b06b8b7fe0000005c` (no response)
-3: `c06b06b8b7fd0000005f`(works)

Just found out now, that it is very likely related to the checksum being 5c, which there is some code to remove duplicates, etc. Will look further into this.

elupus, 2024-12-10:
Yes 5c should be duplicated and/or replaced when its the checksum

elupus, 2024-12-10:
This must be done in the sending app i think. I think we precalculate the block before sending.

https://github.com/yozik04/nibe/blob/cfe4b8173b5fbbcefc9047842111761d9a45d5b7/nibe/connection/nibegw.py#L837

Likely needs the dedupe code of the response type.

qabi, 2024-12-10:
> This must be done in the sending app i think. 

I introduced the `if chksum == 0x5C then chksum = 0xC5`logic, and that fixed it for me.

elupus, 2024-12-10:
Its not fixed yet in neither nibegw code nor nibe python lib so issue is still valid.

elupus, 2024-12-10:
Yes, but we have no issue there yet :)

elupus, 2025-03-26:
Hi. What tool where you using to send data to the device? I just looked at our python project and it already escapes the data and replaces 5c with c5 from what i can tell.

qabi, 2025-03-26:
> Hi. What tool where you using to send data to the device? I just looked at our python project and it already escapes the data and replaces 5c with c5 from what i can tell.

It was a slightly adapted version of another library from github. I don't recall exactly which. I was not using the python project at the time.

So it seems  - in summary - that this was an issue in the code I was using at the time - which I fixed back then. It has been running smoothly since :)

elupus, 2025-03-26:
Ah, then it make sense. Will close this issue. Thanks!

closed 2025-03-26 by elupus