elupus/esphome-nibe #84: Alarm after power outage - race condition?

pzduniak, 2024-12-17:
Hi! I'm incredibly happy to see that you got all the virtual thermostat features working :) Everything seems to be working other than this:

My power supply is not exactly stable due to the power lines running next to trees, so every now and then we get short outages. Yesterday was the first time it happened since getting this project running with my F1145.

I woke up in the morning to a heat pump alarm, reporting that connection with the MODBUS device failed. I think this is due to how we initialize the nibegw module in the ESPHome startup flow:

```cpp
class NibeGwComponent : public esphome::Component, public esphome::uart::UARTDevice {
  float get_setup_priority() const override {
    return setup_priority::BEFORE_CONNECTION;
  }
```

vs

```cpp
namespace setup_priority {
 
 const float BUS = 1000.0f;
 const float IO = 900.0f;
 const float HARDWARE = 800.0f;
 const float DATA = 600.0f;
 const float PROCESSOR = 400.0;
 const float BLUETOOTH = 350.0f;
 const float AFTER_BLUETOOTH = 300.0f;
 const float WIFI = 250.0f;
 const float ETHERNET = 250.0f;
 const float BEFORE_CONNECTION = 220.0f;
 const float AFTER_WIFI = 200.0f;
 const float AFTER_CONNECTION = 100.0f;
 const float LATE = -100.0f;
 
 }  // namespace setup_priority
```

The only special thing about my set up is running a LAN8720 Ethernet module rather than Wi-Fi.

Since all the devices started up at once (network, heat pump, esphome), I think what happened is that ESPHome was stuck waiting for the switches to start up to acquire DHCP, heat pump alarm got triggered and ESPHome finished starting up afterwards.

I think it would make sense to break the dependency of NibeGwComponent on network and move it prior to the network initialization, so that it is guaranteed that we have nibegw initialized soon enough for the heat pump not to alert.

I can make the changes under the maintainers' guidance if needed.

elupus, 2024-12-17:
Sure it would make sense to move it earlier. Maybe even at PROCESSOR.

xHuubx, 2025-02-03:
@pzduniak what hardware are you using? I am looking for a alternative for LilyGO T-CAN485, I would like to run it through a LAN cable rather than WIFI as I also noticed connection issues while using WIFI only.

pzduniak, 2025-02-03:
@xHuubx 

https://www.tindie.com/products/allexok/esp32-stick-eth16mb-flash/
https://www.waveshare.com/ttl-to-rs485-c.htm

Some random buck converter, terminal blocks etc.

And for the race condition I ended up just writing a HA automation to auto-press the reset button when that alert occurs. The temporary solution is slowly becoming the permanent one...

elupus, 2025-03-26:
I've changed priority so it starts up quicker. Should likely work now.

closed 2025-03-26 by elupus