elupus/esphome-nibe #85: can"t connect over modbus tcp to new S1255 HP

scruysberghs, 2024-12-19:
The HP is connected to LAN with cable, ip adres is 192.168.0.232, modbus tcp option is enabled.
I use connect string tcp://192.168.0.232:502 and mobus id 0 to make the connection when configuring the integration. 
Connection fails and in the logs I see:
``` 
File "/usr/local/lib/python3.13/asyncio/selector_events.py", line 681, in _sock_connect_cb
    raise OSError(err, f'Connect call failed {address}')
OSError: [Errno 113] Connect call failed ('19**8** .168.0.232', 502)"
``` 

Looks like a wrong id adress setting is stuck somewhere and is nog being overwritten?

elupus, 2024-12-19:
Does not look like your heatpump is responding. The port 502 is the correct modbus port.

Note this repo is for esp32 based gateway. It has nothing to do with S series pumps with built in modbus. That would be either on https://github.com/yozik04/nibe or home assistant.

But again. Your heatpump is not responding or you have a firewall between your heatpump and the system trying to connect.

closed 2024-12-19 by elupus

scruysberghs, 2024-12-19:
After rebooting home assistant the stuck ip adres is gone, and now I get :
  raise OSError(err, f'Connect call failed {address}')
ConnectionRefusedError: [Errno 111] Connect call failed ('192.168.0.232', 502)

Any tips on what to try to debug further? The ip adres of the heat pump is pingable from Home Assistant. Port 502 is't reported as open but that is expected behaviour for modbus tcp. Is see port 8888 is open on the heat pump ip adres but can't seem te find a service listening there.

scruysberghs, 2024-12-19:
ok sorry, I see this is indeed the wrong repo I'm commenting on. I came here because you are listed as owner on the HA Core Nibe integration. My bad