elupus/esphome-nibe #88: can not update Nibe GW Firmware 2024.12.4

bmwmotor, 2025-01-17:
[logs_nibegw_run.txt](https://github.com/user-attachments/files/18450558/logs_nibegw_run.txt)

It gives an compyling error. Als tried with ESP builder, but same error

elupus, 2025-01-17:
Looks strange. Looks like your host is overloaded or running out of memory. Something kills the linking stage of the build. I dont think this is related to the nibegw firmware.

elupus, 2025-01-17:
Do you have output from esp builder from another machine?

bmwmotor, 2025-01-17:
no, do not have other esp device....It was working fine till the last ESP update this morning. So think it can be a ESP issue ?

bmwmotor, 2025-01-17:
Did restart Home Assistant and after that it updates. Thanks for answering my questions.

closed 2025-01-17 by bmwmotor