elupus/esphome-nibe #89: Ignoring coil XXXXX value XXXXXX - failed to decode

Funky-Fab-4Strings, 2025-02-09:
Hey @elupus , 
using the NibeGW on an LilyGo T-CAN485 since some months now. Basically everything works OK, besides sending Alarm reset from time to time. 
I realized, that since come weeks or months I do not get some of the values. Checked the logs today, and found those messages.
First I relized that i had a Problem with the encryption, which is fixed already. Since the the restart buttons are waorking again. 

Is there anything know with these values. Does anything changed since the move to myUplink? 
Any Idea how to fix this?
Do I need to use the LOG.SET file?

My config:
`esphome:
  name: nibegw
  comment: NibeGw
  platform: ESP32
  board: esp32dev

# General ESPHome setup
api:
  reboot_timeout: 0s
  encryption:
    key: !secret API_KEY

ota:
  - platform: esphome
    password: !secret OTA_PASSWORD

logger:

wifi:
  ssid: !secret WIFI_SSID
  password: !secret WIFI_PASSWORD
  fast_connect: true # Fast connect to connect to my hidden network
  # It is recommended to disable powersave mode on wifi, to make sure the device does not miss UDP requests sent.
  power_save_mode: none
  # The device needs a static IP. Either do that here with the manual_ip node, or do it via another way (router)

# Load nibe component
external_components:
  - source: github://elupus/esphome-nibe

# Set pins required for LilyGo T-CAN485 board
output:
  - platform: gpio
    id: ENABLE_PIN # Enable the chip
    pin:
      number: GPIO19
      inverted: true
  - platform: gpio
    id: SE_PIN # Enable autodirection
    pin:
      number: GPIO17
      inverted: true
  - platform: gpio
    id: ENABLE_5V_PIN # Enable 5V pin for RS485 chip
    pin:
      number: GPIO16
      inverted: true

# Configure uart that will be used
uart:
  rx_pin: GPIO21
  tx_pin: GPIO22
  baud_rate: 9600

# Configure NibeGW
nibegw:
  udp:
    # The target address(s) to send data to. May be a multicast address.
    # When using Home Assistant: this is your Home Assistant IP.
    target:
[... 208 lines cut ...]
2025-02-09 13:42:50.333 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring coil 43420 value 00008a56 - failed to decode
2025-02-09 13:42:51.402 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring coil 43424 value 00009208 - failed to decode
2025-02-09 13:45:47.479 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring coil 43420 value 00008a56 - failed to decode
2025-02-09 13:45:49.314 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring coil 43424 value 00009208 - failed to decode
2025-02-09 13:48:43.254 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring coil 43420 value 00008a56 - failed to decode
2025-02-09 13:48:44.242 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring coil 43424 value 00009208 - failed to decode
2025-02-09 13:51:38.257 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring coil 43420 value 00008b56 - failed to decode
2025-02-09 13:51:39.199 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring coil 43424 value 00009208 - failed to decode
2025-02-09 13:54:38.154 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring coil 43420 value 00008b56 - failed to decode
2025-02-09 13:54:39.164 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring coil 43424 value 00009208 - failed to decode
2025-02-09 13:55:47.132 DEBUG (MainThread) [homeassistant.components.nibe_heatpump] Some coils failed to update, and may be unsupported: Failed to read some or all coils (Failed decoding response for tot-op-time-compr-eb100-ep14-43420: Failed to decode tot-op-time-compr-eb100-ep14-43420 coil from raw: 00008b56, exception: Raw value 1451950080 is out of range for coil tot-op-time-compr-eb100-ep14-43420, Failed decoding response for tot-hw-op-time-compr-eb100-ep14-43424: Failed to decode tot-hw-op-time-compr-eb100-ep14-43424 coil from raw: 00009208, exception: Raw value 143785984 is out of range for coil tot-hw-op-time-compr-eb100-ep14-43424)
2025-02-09 13:55:47.133 DEBUG (MainThread) [homeassistant.components.nibe_heatpump] Finished fetching Nibe Heat Pump data in 117.748 seconds (success: True)
2025-02-09 13:57:30.102 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring coil 43420 value 00008b56 - failed to decode
2025-02-09 13:57:31.116 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring coil 43424 value 00009208 - failed to decode
2025-02-09 13:58:44.080 DEBUG (MainThread) [homeassistant.components.nibe_heatpump] Some coils failed to update, and may be unsupported: Failed to read some or all coils (Failed decoding response for tot-op-time-compr-eb100-ep14-43420: Failed to decode tot-op-time-compr-eb100-ep14-43420 coil from raw: 00008b56, exception: Raw value 1451950080 is out of range for coil tot-op-time-compr-eb100-ep14-43420, Failed decoding response for tot-hw-op-time-compr-eb100-ep14-43424: Failed to decode tot-hw-op-time-compr-eb100-ep14-43424 coil from raw: 00009208, exception: Raw value 143785984 is out of range for coil tot-hw-op-time-compr-eb100-ep14-43424)
2025-02-09 13:58:44.080 DEBUG (MainThread) [homeassistant.components.nibe_heatpump] Finished fetching Nibe Heat Pump data in 116.695 seconds (success: True)

2025-02-09 14:00:32.049 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring coil 43420 value 00008b56 - failed to decode
2025-02-09 14:00:33.048 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring coil 43424 value 00009208 - failed to decode
2025-02-09 14:01:43.031 DEBUG (MainThread) [homeassistant.components.nibe_heatpump] Some coils failed to update, and may be unsupported: Failed to read some or all coils (Failed decoding response for tot-op-time-compr-eb100-ep14-43420: Failed to decode tot-op-time-compr-eb100-ep14-43420 coil from raw: 00008b56, exception: Raw value 1451950080 is out of range for coil tot-op-time-compr-eb100-ep14-43420, Failed decoding response for tot-hw-op-time-compr-eb100-ep14-43424: Failed to decode tot-hw-op-time-compr-eb100-ep14-43424 coil from raw: 00009208, exception: Raw value 143785984 is out of range for coil tot-hw-op-time-compr-eb100-ep14-43424)
2025-02-09 14:01:43.031 DEBUG (MainThread) [homeassistant.components.nibe_heatpump] Finished fetching Nibe Heat Pump data in 119.647 seconds (success: True)
2025-02-09 14:03:25.999 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring coil 43420 value 00008b56 - failed to decode
2025-02-09 14:03:26.996 WARNING (MainThread) [nibe.nibe.connection.nibegw] Ignoring coil 43424 value 00009208 - failed to decode
2025-02-09 14:04:37.988 DEBUG (MainThread) [homeassistant.components.nibe_heatpump] Some coils failed to update, and may be unsupported: Failed to read some or all coils (Failed decoding response for tot-op-time-compr-eb100-ep14-43420: Failed to decode tot-op-time-compr-eb100-ep14-43420 coil from raw: 00008b56, exception: Raw value 1451950080 is out of range for coil tot-op-time-compr-eb100-ep14-43420, Failed decoding response for tot-hw-op-time-compr-eb100-ep14-43424: Failed to decode tot-hw-op-time-compr-eb100-ep14-43424 coil from raw: 00009208, exception: Raw value 143785984 is out of range for coil tot-hw-op-time-compr-eb100-ep14-43424)
2025-02-09 14:04:37.988 DEBUG (MainThread) [homeassistant.components.nibe_heatpump] Finished fetching Nibe Heat Pump data in 115.603 seconds (success: True)`

elupus, 2025-02-09:
Check if the wordswap setting in your pump have changed. Have seen that happen on firmware updates of the pump.

Funky-Fab-4Strings, 2025-02-09:
It was turned off (no hook inside) is this supposed to be on? 
Actually read the first time about it.

closed 2025-05-28 by elupus