elupus/esphome-nibe #95: Question: BF2 Flowmeter Data Availability via Modbus?

tobbensoft, 2025-02-26:
I want to clarify something regarding the Modbus data exposure in my system.

**System Configuration**

- Components: VVM 500 + F2120-20 + 2 Extra Climate Systems (ECS 41)
- Flowmeters:
  - BF1: Connected to X22 and working (sensor.bf1_ep14_flow_40072 in Home Assistant)
  - BF2: Connected to X23 (on heating return from house), but no flow entity appears for it
- Integration: Using LilyGo T-CAN485 to bring data into Home Assistant
- Additional Notes: Cooling and Pool to be installed

**Issue**
All expected entities appear in Home Assistant, and BF1 (sensor.bf1_ep14_flow_40072) works as expected. However, BF2 does not seem to be exposed as a flowmeter in Modbus.

Looking at the official documentation: [Modbus S-Series Documentation] https://www.nibe.eu/download/18.3db69dc1795e0d992c5722/1622634529178/Modbus%20S-series%20EN%20M12676EN-1.pdf , it appears that BF2 is not exposed via Modbus.

**Question**
Is my assumption correct that Nibe does not expose BF2 flowmeter data on Modbus, or am I missing something?

Any insights would be greatly appreciated!

Thanks!

elupus, 2025-02-26:
I suspect BF2 may be available. But we dont know the parameter number for it. You could ask nibe for the modbus40 parameter for this value and see if you get a reply.

elupus, 2025-03-23:
Closing this for now. If you did get some response would be nice to get it added to the nibe lib with the parameter databases.

closed 2025-03-23 by elupus