elupus/esphome-nibe #96: How to use with actual RMU40

AbsolutJohan, 2025-03-22:
So I have the Lilygo t-can485 ready but my NIBE 1145 already have a RMU40 connected to it. The RMU 40 uses the same port on the board in Nibe heat pump (9, 10, 11, 12). I haven't enabled the dummy RMU40 accessory in config.

Is it okay to connect it to the same plints? Or will this not work, anyone with the same setup? Asking first so I don't destroy anything and is left with a frozen house

elupus, 2025-03-22:
It should work on same pins. But can depend on how long cabling you have. You can look at the install instructions for modbus40. Normally they recommend to a avoid star connections. But if you only have one rmu its likely to work.

AbsolutJohan, 2025-03-23:
Thanks! So now I have plugged it in and it does work together with the RMU40 🙂 thank you for your awesome work on this 🎉

elupus, 2025-03-23:
Nice to hear! Had a few people who had problems getting to work. So nice with some success stories as well :)

closed 2025-03-23 by elupus