elupus/esphome-nibe #98: Failed to perform the action climate/set_temperature when changing temperature

olivebadger14, 2025-05-11:
home assistant 2025.4.4

when I change the temperature for my heat pump through Home Assistant, if I do it too quickly, I get the attached the temperature does not change.

sometimes the speed that I’m making the change doesn’t matter I still get this error the first time I change the temperature eventually the temperature will change if I keep doing it

[image]

the below is in the home assistant logs

Logger: nibe.nibe.connection.nibegw
Source: runner.py:154
First occurred: 09:31:11 (1 occurrences)
Last logged: 09:31:11

Ignoring packet from ('192.168.1.171', 9999) due to parse error: 5c007e2d4b5c00206b004b5c00206900495c00206900495c00206900495c41c9880000065c41c9890001065c41c96500ed065c41c98c020080865c41c9550100dc5c41c98b1480109c008110890083107e: Error in path (parsing) -> data -> checksum wrong checksum, read 126, computed 142

elupus, 2025-05-11:
Those type of parse errors usually indicate something wrong with the connection to the pump from the ESP GW device.

The denied writing error is weird.. we have seen those when writing alarm, but mostly not other places.

olivebadger14, 2025-05-12:
I have been testing this a bit more and to make things more interesting, if i change the temperature to over 25 c (heat pump goes up to 30c) the error does not appear but the temperature toggle disappears and my home assistant card just shows heat rather than the selected temp

olivebadger14, 2025-05-12:
> Those type of parse errors usually indicate something wrong with the connection to the pump from the ESP GW device.
> 
> The denied writing error is weird.. we have seen those when writing alarm, but mostly not other places.

i ll keep trying to debug, see if my yaml config has any errors.

Tommatheussen, 2025-05-14:
I am also occasionally having write errors lately, mainly when I try to enable cooling mode (`allow-cooling-47372`) on my heatpump. I've updated and connected the heatpump to MyUplink (was connected to NibeUplink before) and it's running the latest version, not sure if it's related to the latest firmware on the heatpump or not...

tRoOlos, 2025-05-15:
I've the same error using entity climate.climate_system_s1 (Nibe F1145 heatpump).

Also the following: 
On the Heatpump and also on RMU40 I have the following options to set the heatpump mode:
1. Auto
2. Manual: with sub options: addition, heating or cooling
3. Add. heat only:  with sub options: heating or cooling

[image]

[image]

[image]

**The limate.climate_system_s1  entity gives me the options:**
1. Auto
2. Heating/cooling -> but when I select it I cannot set it to or heating or cooling or addition
3. Heating -> I cannot choose heating or cooling

[image]

**So I cannot set the heatpump to manual and then the option: cooling (for example)? Like I can on the heatpump or the Nibe RMU40. Is this by design?**

When I change the operating mode on the climate.climate_system_s1 I get mostly the error. It doesn't actually change the modes (only on the card).

**Errors**:

[image]

[image]

Also when using the entity: select.operational_mode_47137 I can only select the operational mode of the heat pump 0=Auto 1=Manual 2=Add. heat only. There is no option to manual select: cooling for example. 
Is there a way wit a modbus write register to directly set the heatpump manual in Heating or Cooling mode?

[image]

Thanks for all your effort and help.

elupus, 2025-05-28:
For the manual mode you would need to adjust some of the other entities. I've not looked how to map this to same logic as is displayed on the pump. But nice to see some screenshots on how it looks on the pump.

Might be possible to achieve something similar. 

Regarding the error with writing. This started happening with some firmware update of the pumps a while back. We still don't know why that gives errors. The pump is rejected our attempts to set the parameters.

elupus, 2025-05-30:
The weird behaviour with 25.5 volt seem to be due to RMU data limitations. Can you set higher than 25.5 degrees target set point on your actual RMU?

When set point goes over 25.5 we get weird data over the RMU data line. This breaks the normal reading.