elupus/esphome-nibe #99: Unexpected Ack/Nack: 5C when setting up Nibe F1245

xwl, 2025-05-18:
Hi,

So in order to be able to control the Nibe heat pump, I purchased a LilyGO T-CAN485 and did the cabling and configuration mostly similar to the examples here:

https://github.com/elupus/esphome-nibe/blob/master/examples/lilygo-t-can485.png

I got lots of these error, also Nibe alarms: communication error ...

    [19:57:14][D][nibegw:106]: Response to address: 0x20 token: 0xee bytes: 7
    [19:57:14][D][uart_debug:114]: <<< 5C:00:20:EE:00:CE
    [19:57:14][D][uart_debug:114]: >>> C0:EE:03:0A
    [19:57:14][D][uart_debug:114]: >>> 00:01:26
    [19:57:15][W][nibeGW:115]: Unexpected Ack/Nack: 5C
    [19:57:15][D][nibegw:043]: UDP Packet with 6 bytes to send

It sounds like the protocol mismatches? Do you see any issue here? 

Full log:

[logs_nibe-f1245_logs.txt](https://github.com/user-attachments/files/20274205/logs_nibe-f1245_logs.txt)

My config, which is based on the example config:

[nibe-f1245.yaml.txt](https://github.com/user-attachments/files/20274231/nibe-f1245.yaml.txt)

Cabling:

[image]

elupus, 2025-05-18:
The transmission is not reaching the pump. So it is not acknowledging that we sent anything.

Your tx enablement could be incorrect in your config maybe.

elupus, 2025-05-18:
Hmm. Config looks correct. I wonder if you might need to add a ground cable on the signal side on the rs485 connector. It "should" not be needed, but given that i've seen other similar reports with these boards. Maybe the nibe boards have some weird way of handling the differential signal.

elupus, 2025-05-18:
The other possible cause could be termination issue. That you need 120ohm resistor between the A and B terminals.

xwl, 2025-05-19:
> Hmm. Config looks correct. I wonder if you might need to add a ground cable on the signal side on the rs485 connector. It "should" not be needed, but given that i've seen other similar reports with these boards. Maybe the nibe boards have some weird way of handling the differential signal.

If in addition I connected GND at port 9 to the rs485, I don't get any UDP messages output at all.

xwl, 2025-05-19:
> That is utterly weird.
> 
> See if you can get a 120ohm resistor to place between A nd B to terminate the line correctly. I would expect the pump to be terminated so i dont expect it to help, but maybe.

No, I was wrong. It was cabling issue, wasn't good connected. Now after I reconnect the cable, your first suggesting, **connecting the GND also to rs485** actually fixes my issues. :)

Thanks for your help!

closed 2025-05-19 by xwl