elupus/esphome-nibe #105: Modbus alarms with v3.1.2

gerbenvl, 2025-05-30:
Getting modbus alarms all the time after updating to to v3.1.2. Resetting the alarm doesn't help, the alarms come back quickly. All data is still arriving in HA though, but the alarms are quite annoying. Running ESPHome 2025.5.1 with a lilygo-t-can485 connected to a Nibe SMO40. The ESP log shows:

`[08:17:29][W][nibeGW:224]: Had invalid message`

Rolled back to @v3.1.1 for now, no alarms with that versions.

elupus, 2025-05-30:
Can you try master?

elupus, 2025-05-30:
Hmm. Maybe that is what you did?. 3.1.2 har nothing that should affect that, but master does.

gerbenvl, 2025-05-30:
Yes that's what I did. I assumed that the latest version equaled the master branch :-) I will update the title.

3.1.2 is indeed fine, but the problem is on the current master.

elupus, 2025-05-30:
Ok good. Thin you can turn on debug level logs for me. Or even verbose level logs and post them here? I would very much like to track down the issue so i dont have to revert stuff.

gerbenvl, 2025-05-30:
> Ps. I think [b405259](https://github.com/elupus/esphome-nibe/commit/b405259f9a0cae72b5ccfa46607ec1d1b802f3fc) should be fine, but would be nice with confirmation too.

I installed like this: "source: github://elupus/esphome-nibe@b405259", but that unfortunately gives the alarm problem.

> Ok good. Think you can turn on debug level logs for me. Or even verbose level logs and post them here? I would very much like to track down the issue so i dont have to revert stuff.

Sure, if you have some instructions how do do it / what you need? Or is this ok:

[logs_lilygo-t-can485_run.txt](https://github.com/user-attachments/files/20518282/logs_lilygo-t-can485_run.txt)

This log is with the version from the current master branch btw.

elupus, 2025-05-30:
Yup, that gave me enough details to find a bug. Could you try master again? (turn down debug levels to start with since that can interfere)

gerbenvl, 2025-05-30:
Didn't seem to have worked. Indeed ran with default debug level. Ran again now with verbose to make a log: 

[logs_lilygo-t-can485_run 12.27 30-05-2025.txt](https://github.com/user-attachments/files/20519003/logs_lilygo-t-can485_run.12.27.30-05-2025.txt)

elupus, 2025-05-30:
Could you check with explicitly grabbing https://github.com/elupus/esphome-nibe/commit/f3a1bbc7b0ea98a3ec1c177e09ec73770a87bcf4 so it's not using some cached version of master before the change?