elupus/esphome-nibe #107: ERROR: async_udp

buuenke, 2025-05-30:
Integration worked fine.
Now not anymore...dont know since when.

I get strange error messages:
[23:33:34][W][nibegw:071][async_udp]: UDP Packet wrong ip ignored 192.168.1.14
[23:33:35][W][nibegw:071][async_udp]: UDP Packet wrong ip ignored 192.168.1.28
[23:33:50][W][nibegw:071][async_udp]: UDP Packet wrong ip ignored 192.168.1.21

What to do??

elupus, 2025-05-30:
Those warnings from other devices on your network sending junk. The default port is problematic. But it should still work.

You might want to test a clean rebuild. If you rebuilt the device sometime last two days , there was some bugs.

elupus, 2025-05-30:
We need to get logging turned up a bit. Debug or verbose.

elupus, 2025-05-30:
Wait. If you had a firmware update. I would guess it turned off the modbus accessory in the pump.

buuenke, 2025-05-31:
I checked already. The box is checked at service menu 5.x.x.

Yahoo Mail: Zoeken, organiseren, veroveren

elupus, 2025-06-01:
Its not getting any data from the pump from what i can tell. So if modbus40 is enabled in your pump, the pump should go into alarm. If its not going into alarm, i dont think modbus 40 is enabled.

Maybe try disabling it in the pump, and reenable it.

elupus, 2025-06-01:
If it does go into alarm, i would check your cabling.

buuenke, 2025-06-01:
Yes I goed into alarm all te time 

Yahoo Mail: Zoeken, organiseren, veroveren

buuenke, 2025-06-07:
I even bought a new lily board...
May has to do something with upgrading firmware of NIBE....?

elupus, 2025-07-07:
Make sure you remove any non needed features from your config. Seems you have a webserver active for example.