elupus/esphome-nibe #108: FSM does not expect data response after request with len != 0 and enters wrong state?

nptr, 2025-06-08:
Hi,

i am trying to get the gateway running (just as MODBUS40 device), but was unsuccessful. My VVM goes into alarm state after ~10 seconds.
When inspecting and analysing the traffic, I came across messages that the protocol parser FSM does not expect:

**Observation:**

```
Unexpected Ack/Nack: C0
Recv: 5F 41 C9 34 01 00 BD C0
Ignoring byte 34
Ignoring byte 01
Ignoring byte 00
Ignoring byte F5
Ignoring byte 06
```

Master's request `5c 41 c9 34 01 00 bd` has a valid checksum (xor 41 c9 34 01 00 = bd)
Slave's response `c0 34 01 00 f5 06` has a valid checksum (xor c0 34 01 00 = f5)

**Possible error:**
The problem(*) seems to be in [`NibeGw::handleMsgReceived():L208`](https://github.com/elupus/esphome-nibe/blob/master/components/nibegw/NibeGw.cpp#L208).
The FSM only expects a long response (next state `STATE_WAIT_START_SLAVE`) , when the master's `len == 0`.
This assumption seems to be false. There can be a long response, even if master sent data.  Maybe a parameterized request for data.

(*) Not too much of a problem, FSM just goes into STATE_WAIT_START earlier than it should?

**Possible Solution:**
You could remove the check and unconditionally transition into `STATE_WAIT_START_SLAVE` because it [already handles the case](https://github.com/elupus/esphome-nibe/blob/master/components/nibegw/NibeGw.cpp#L108).

**Conclusio:**
Note that this did not solve my problem. The parser still sees **other** telegrams it does not expect, but I haven't verified yet whether the parser is at fault or if those are indeed garbage. My hope is, that the FSM is just confused by those telegrams and once that is resolved, everything works.

Let me know what you think. And please excuse me not creating a PR for this, I am not deep enough into this project yet.

elupus, 2025-06-08:
I rewrote this parser a little back. The new parser tries to parse slave responses, old one did not. You can try the last tagged version and see if that works better. But would be nice to see this solved.

Ive never seen any slave responses when length is non zero. But i agree your data does indicate that to be a possible case. We would need some slave start wait that always goes active after a master message, but falls back to normal logic if  no slave start is seen.

The 06, after a slave response i have interpreted as an ACK from the master on the data sent from the slave.

nptr, 2025-06-14:
Update: Apart from the issue reported, the software works properly as far as i can tell. 

The reason I can't get it to work must be in hardware somehow. Reading directly from the bus with my PC, I can see that no acknowledges are sent, even though the logs say so. Communicating with just my PC (simulating the heatpump) via USB<>RS485 there are acknowledges however. 

For reference, I am using the M5 StamPLC. I tried various timings for enabling/disabling the driver. I tried with and without the 120Ω terminating resistors. I also tried with and without the bias resistors. Maybe using some other hardware will be more successful...

elupus, 2025-06-14:
That sound very much like your config is not enabling TX on the transceiver when its sending.

nptr, 2025-06-14:
I checked that.

- When nibegw + heatpump + PC (listening) on a bus => no acknowledges received by PC (and heatpump) => alarm state
- When nibegw + heatpump on a bus => probably no acknowledges -> alarm state
- When nibegw + PC on a bus (simulating the pump) => acknowledges are there

So the combination of my hardware and heatpump seems problematic.
Gotta check the signal quality with a scope or try a different hardware altogether...

nptr, 2025-06-14:
Yes. And I made sure to disable UART logging.

nptr, 2025-06-16:
I had a look at it with a logic analyzer/MSO. Took me a while to see it, but the acknowledges are there. They are just super late (100+ ms), often colliding with other messages and are not recognizable anymore.

Not sure if there suddenly is buffering, or ESPHome/my config just make things that indeterministic. It seemed to have worked for you and others.. Removing all logging, the 1ms delays and forcing the high frequency loop to be active at all times didn't help. I'll see if maybe removing some peripherals from [my config (based on this)](https://github.com/Beormund/esphome-m5stamplc/tree/main) has an effect. It is comparatively large and i have no idea what ESPHome does with it. Last resort, i'll write my own firmware.

elupus, 2025-06-17:
It really only should act as gareway  nothing else. Thinking is critical.

elupus, 2025-07-07:
Ah nice! I wish we could figure out why esphome messed up the response timing for you. There must be something blocking the loop.

elupus, 2025-07-07:
Do you have the minimal config you used while still getting issues? Could maybe help me track it down.

nptr, 2025-07-08:
I checked, but no. I seem to have reverted the changes after the unsuccessful attempts. Config is the one i've linked earlier.
Also I seem to have closed the issue accidentally yesterday, I'll reopen it for its original purpose - to keep note of the FSM bug.

nptr, 2025-12-15:
Life is keeping me busy currently and I might not have enough blocked time to cleanly switch between mine and your solution in a sitting, as to not risk prolonged downtime of my monitoring this heating period. Maybe come summer, or when opportunity arises.