elupus/esphome-nibe #112: Esphome 2025.8. framework

Jirik6, 2025-08-22:
Hi,
from esphome 2025.8.0 is warning to set framework. Recommended option to ESP-IDF not work. Only arduiono opt.

Thx. J.

elupus, 2025-12-14:
This is now resolved.

elupus, 2025-12-14:
Do you have log from your esp? and your config?

Jirik6, 2025-12-14:
and log from esphome is unusable cannot connect to.... now with 2025.2 is everthing fast, ping imiditely, webserver imidietly...

elupus, 2025-12-14:
Was that with arduino or espidf as platform?

elupus, 2025-12-14:
And look at the log

Jirik6, 2025-12-16:
> If you just run in locally at your pc and grab log output from it there. How does that look? It should still connect to wifi and you should be able to ping it

Pls close it I dont have time to solve it. Stay on 2025.2. Thx