elupus/esphome-nibe #115: would it be possible to add emulation for SMO 40

Arut42, 2025-09-26:
Hello, would it be possible to support smo40?

I would try it myself, but I don't understand the connections between esphome and the C code of nibegw.

I mainly work with Arduino :)

Arut42, 2025-09-27:
sry my bad, did not verify gpt answere

i mean de DEH500 with bus adress 0x0027

Arut42, 2025-09-27:
i tryed it, but then i get to much info and the esp crashes as soon as i open the log.

Arut42, 2025-09-27:
Got the debuging to run what i found until now:

new tokens
ECS_DATA_REQ=		0x90,
ECS_DATA_MSG_1=	0x55,
ECS_DATA_MSG_2=	0xA0,

request with address 27
Recv: 5C 00 27 90 00 B7 06 
Recv: 5C 00 27 55 02 00 00 70 06 

i found a similar request from some internal module of the heatpump itself
Recv: 5C 00 2B    90     00 BB C0 90 10 FF 03 FF 03 B6 02 FF 03 FF 03 FF 03 FF 03 FF 03 08 06 
after request, this msg follow
Recv: 5C 00 2B   A0   02 64 00 ED 06  <= answer msg 1
Recv: 5C 00 2B   55   02 00 00 7C 06  <= answer msg 2

so what i need to do is create the missing A0 token msg :)

these are fully ident like time is over for the answer
Recv: 5C 00 27 55 02 00 00 70 06 
Recv: 5C 00 2B 55 02 00 00 7C 06

elupus, 2025-09-28:
Add the ids of know unit here if https://github.com/elupus/esphome-nibe/blob/ea0df5abbdf9228501218c28928306a1fd1c78cf/components/nibegw/NibeGw.h#L87 if you get a chance. Always good to get that list more complete.

Note we have another list here: https://github.com/yozik04/nibe/blob/340f4713665e5776d7a2321055cd19ab689968e7/nibe/connection/nibegw.py#L715

In general parsing the data is generally easier to do in the python lib. The espunit will forward all packets it gets over udp. Somewhat harder to do responses.

Arut42, 2025-09-29:
Ok i get all frames :) just wrong formatet in my log

5C 41 C9 73 00 FB 
C0 73 09 FC 04 00 00 00 26 E7 2B 0F A7 
06 
5C 41 C9 ED 00 65 
C0 ED 18 7D 31 00 00 B2 5A 10 00 11 47 00 00 AB E5 0F 00 6A 00 00 00 01 00 00 00 FD 
06 
5C 41 C9 B7 00 3F 
C0 B7 18 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00 D4 00 00 00 00 00 F1 00 48 
06 
5C 41 C9 F2 00 7A 
C0 F2 22 00 00 FF FF 00 00 00 00 22 01 7C 01 00 00 00 00 00 00 00 00 98 08 55 13 10 00 00 50 00 00 00 00 00 00 D8 
06 
5C 41 C9 F1 00 79 
C0 F1 26 8E 00 96 00 D3 00 A4 00 88 00 C7 00 8C 00 6E 00 88 00 6C 00 80 00 C6 00 6D 00 85 00 AF 00 00 00 0A 00 41 19 37 0A 5F 
06 
5C 41 C9 A0 02 00 00 2A 
06 
5C 41 C9 F7 00 7F 
C0 F7 20 50 00 3F 00 EF FF 8D 00 00 00 88 00 00 00 00 00 5A 00 41 00 E7 FF 84 00 00 00 85 00 85 00 00 00 EA 
06 
5C 41 C9 FC 00 74 
C0 FC 13 00 00 00 06 FF 00 06 FF 00 06 FF 00 00 CE FF 00 CE FF 00 D6 
06

elupus, 2025-09-29:
Nice! Since I've not fully followed :). What is it you are trying to accomplish? Replacing a physical DEH500 with some virtual one controlled by the esp/ha?

Arut42, 2025-10-03:
Yes, I have a water-operated stove that is connected to the vvm.
This heats the hot water, but sometimes it uses too much energy and the water temperature rises above 65°C.
A mixer (qn10) is installed in the vvm, which can mix the excess hot water into the heating circuit.
However, this only becomes active if a deh500 is present.

final findings
for the Version ident   // Version 1000 = 03E8 => E8 03  maybe the 2 bytes are enougth
 request_data_type payload = {0xE8,0x03,0x00,0x00,0x00,0x00,0x00};  // Version 1000 = 03E8 => E8 03 
 request_data_type data = { STARTBYTE_SLAVE, ACCESSORY_TOKEN, (byte) payload.size() };  

for the cyclic data the 2 Byte for the bt52 are enough too remove the error start the controlling
request_data_type payload = { Byte1 , Byte2 }; //0x00, 0x00, 0x00, 0x00, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00
request_data_type data = { STARTBYTE_SLAVE, ECS_DATA_REQ, (byte) payload.size() }; 

The temp decoding is a little bit strange

<html>
<head>
	
	
	<title></title>
	<meta name="generator" content="LibreOffice 7.3.2.2 (Windows)"/>
	<style type="text/css">
		body,div,table,thead,tbody,tfoot,tr,th,td,p { font-family:"Liberation Sans"; font-size:x-small }
		a.comment-indicator:hover + comment { background:#ffd; position:absolute; display:block; border:1px solid black; padding:0.5em;  } 
		a.comment-indicator { background:red; display:inline-block; border:1px solid black; width:0.5em; height:0.5em;  } 
		comment { display:none;  } 
	</style>
	
</head>

<body>

Byte0 | Byte1 | Temp
-- | -- | --
0 | 0 | err
0 | 24 | err
0 | 25 | 169,1
0 | 50 | 138,1
0 | 86 | 115,7
0 | 154 | 92,8
0 | 181 | 86,6
0 | 255 | 73,2
1 | 0 | 73,1
1 | 50 | 65,9
1 | 100 | 59,6
1 | 150 | 53,9
1 | 200 | 48,6
1 | 255 | 43,1
2 | 220 | 21,3
3 | 128 | 0,1
3 | 129 | -0,1
4 | 255 | -12,8
4 | 200 | -19,4
4 | 156 | -25,9
4 | 128 | -31
4 | 83 | -43,9
4 | 69 | -46,6
4 | 62 | -49,4
4 | 61 | err

</body>
</html>

if the external temp (bt52) is higher than the vvm hotwater it switch on the external circulation pump.

after the temp in the vvm rises above 3°C over the "hotwater stop temp" it should open the qn10.
I will test this the next days :)

Arut42, 2025-10-06:
i have already finished it